vvuk/eddy-ng #2: Error: min height for valid samples is too high

piotrb, 2025-02-07:
Hi,

Trying to get this to work with my Eddy Coil on my Trident .. running into this error during calibration:

I do the manual calibration bit, and when I press Accept this error happens.

```
Error: min height for valid samples is too high: 1.832 > 0.65. Refer to the documentation for troubleshooting.
Calibration failed
```

I removed all my "normal" eddy stuff, and just started with a clean config .. 

Here is the csv in case you need it ..

[eddy-calibration.csv](https://github.com/user-attachments/files/18712113/eddy-calibration.csv)

Not a lot special in my cfg .. 

[eddy-ng.cfg.txt](https://github.com/user-attachments/files/18712150/eddy-ng.cfg.txt)

piotrb, 2025-02-07:
This was on: klipper v0.12.0-410-gcf3b0475d-dirty .. 

I'm updating to v0.12.0-432-gfec3e685c just in case right now

suprs0nic, 2025-02-07:
I just set up my Eddy can on my idex and had the same issue. After adjusting the height of Eddy to around 2.5-3mm above nozzle the error was gone. Before Eddy has around 1.5mm above the nozzle and i recieved the same error with "1.2 > 0.65"

vvuk, 2025-02-08:
Yep, what @suprs0nic said -- sounds like it's too low.

piotrb, 2025-02-08:
needs a troubleshooting entry in the wiki :)

closed 2025-02-20 by vvuk