vvuk/eddy-ng #3: !! Error during homing z: Probe did not clear safe time threshold.

Zinput, 2025-02-09:
Homing z seems to be unreliable for me. Sometimes, it just stops homing z almost instantly even thought it's one cm or two above the bed. Other times, it succeeds. Also, I am unable to test if Z=0 is good because I am unable to actually move the height to Z=0. It always sets it to 0 plus the z-offset calculated by tap or just some random value close to but greater than 0, when I enter for it to move to 0 on absolute positioning in Fluidd.

vvuk, 2025-02-09:
This sounds like bad calibration/bad drive current. Can you retry via `PROBE_EDDY_NG_SETUP` and post the results here, as well as the contents of /tmp/eddy-calibration.csv after running setup? (If you don't have _SETUP, update from the repo, it showed up today)

Zinput, 2025-02-11:
Results of running `_SETUP`
```
// EDDYng setup: calibrating homing
18:34:40  // Drive current 16: valid height: 1.091 to 14.970, freq spread 1.69% (3201798.5 - 3255759.1), Fit 0.0148 (17.13)
18:34:47  // Drive current 17: valid height: 0.029 to 14.977, freq spread 3.15% (3187017.5 - 3287352.6), Fit 0.0109 (17.72)
18:34:47  // EDDYng using 17 for homing.
18:34:54  // Drive current 17: valid height: 0.025 to 14.973, freq spread 3.16% (3186994.1 - 3287583.7), Fit 0.0121 (17.51)
18:34:54  // EDDYng using 17 for tap.
18:34:54  // EDDYng setup success. Please check whether homing works with G28 Z, then check if tap works with PROBE_EDDY_NG_TAP.
18:34:54  // Calibration saved. Issue a SAVE_CONFIG to write the values to your config file and restart Klipper.
```
Contents of `/tmp/eddy-calibration.csv`
```
time,frequency,avg_freq,z,avg_z
989.513031,3187030.882,3187011.0445,15.0,14.974999999999998
989.515049,3187030.882,3187011.53,15.0,14.970138888888888
989.517067,3187015.414,3187011.9184,14.993749999999999,14.964999999999998
989.519086,3187015.414,3187019.6326363636,14.981249999999996,14.96022727272727
989.521104,3187065.84,3187005.3820833336,14.96875,14.955208333333331
989.523122,3187081.307,3187011.222461539,14.962499999999999,14.950480769230769
989.525141,3186833.203,3187016.2285,14.949999999999996,14.945535714285713
989.527159,3187015.414,3187004.0268,14.943750000000001,14.940416666666666
989.529177,3187015.414,3187006.357333333,14.931249999999999,14.93125
989.531195,3187015.414,3187009.7189999996,14.918749999999996,14.921249999999999
989.533214,3187096.775,3187014.111866666,14.912500000000001,14.910833333333334
989.535232,3186848.626,3187018.504733333,14.899999999999999,14.900833333333335
989.53725,3187081.307,3187019.5358666657,14.893750000000004,14.890833333333335
989.539269,3187081.307,3187019.5358666657,14.881250000000001,14.880833333333333
989.541287,3186833.203,3187036.0761333327,14.868749999999999,14.870833333333335
989.543305,3187065.84,3187040.4689999986,14.862500000000004,14.86041666666667
989.545324,3187081.307,3187044.861866666,14.850000000000001,14.850416666666668
989.547342,3187081.307,3187049.254733332,14.837499999999999,14.840416666666668
989.549359,3187081.307,3187048.2235333323,14.831250000000004,14.830416666666668
989.551376,3187081.307,3187063.735599999,14.818750000000001,14.820416666666667
989.553393,3187081.307,3187080.2759333323,14.8125,14.810000000000002
989.55541,3187081.307,3187084.668799999,14.800000000000004,14.8
989.557427,3187081.307,3187105.601933333,14.787500000000001,14.790000000000001
989.559444,3187081.307,3187094.4856666676,14.78125,14.779583333333335
989.561461,3187081.307,3187094.4856666676,14.768750000000004,14.769583333333335
989.563479,3187081.307,3187098.8785333335,14.762499999999996,14.759583333333333
989.565496,3187081.307,3187102.240266667,14.75,14.749583333333334
989.567513,3187329.412,3187118.7805999997,14.737500000000004,14.739583333333334
989.56953,3187147.2,3187122.142333334,14.731249999999996,14.729166666666666
989.571547,3187147.2,3187108.9637333336,14.71875,14.719166666666666
989.573564,3186899.096,3187113.3566000005,14.706250000000004,14.709166666666667
989.575581,3187081.307,3187118.7806666666,14.699999999999996,14.699166666666665
989.577598,3187147.2,3187123.1735333335,14.6875,14.689166666666665
989.579615,3187131.733,3187124.2047333336,14.681249999999999,14.678749999999999
989.581632,3187329.412,3187145.1378666665,14.668749999999996,14.668749999999998
989.583649,3187131.733,3187132.9904,14.65625,14.658749999999998
989.585666,3186883.628,3187132.9904000005,14.649999999999999,14.648333333333332
989.587684,3187147.2,3187132.9904000005,14.637499999999996,14.63833333333333
989.589701,3187162.668,3187148.499533334,14.631250000000001,14.62833333333333
989.591718,3187147.2,3187151.8612666675,14.618749999999999,14.618333333333332
989.593735,3187096.775,3187150.8301333343,14.606249999999996,14.60833333333333
989.595752,3187395.304,3187152.892466667,14.600000000000001,14.597916666666666
989.597769,3187147.2,3187141.7761999997,14.587499999999999,14.587916666666667
989.599786,3187147.2,3187142.8073333334,14.574999999999996,14.577916666666669
989.601803,3187147.2,3187160.3788,14.568750000000001,14.567916666666667
989.60382,3187131.733,3187164.771666667,14.556249999999999,14.557916666666667
989.605837,3187131.733,3187168.133333334,14.550000000000004,14.547500000000001
989.607854,3187131.733,3187171.4950666674,14.537500000000001,14.5375
989.609872,3187162.668,3187179.2496000007,14.524999999999999,14.527500000000002
989.611889,3187162.668,3187167.1022000005,14.518750000000004,14.517500000000002
989.613906,3187147.2,3187171.495066667,14.506250000000001,14.5075
989.615923,3187147.2,3187159.347666667,14.5,14.497500000000002
989.61794,3187213.093,3187163.740533334,14.487500000000004,14.4875
989.619957,3187213.093,3187164.771666667,14.475000000000001,14.477500000000001
989.621974,3187197.626,3187170.1956666675,14.46875,14.467083333333335
989.623991,3187213.093,3187188.8012666674,14.456250000000004,14.457083333333335
[... 1411 lines cut ...]
992.470142,3278411.91,3278165.3018666664,0.22500000000000142,0.22625000000000076
992.47216,3278660.059,3278595.4089333336,0.21875,0.21625000000000133
992.474177,3279043.794,3279022.4792,0.20625000000000426,0.20625000000000096
992.476194,3279540.092,3279475.024466667,0.19375000000000142,0.19625000000000103
992.478211,3279968.575,3279920.095333333,0.1875,0.18583333333333438
[... cut ...]
992.504436,3285924.017,3285637.749071428,0.05624999999999858,0.06026785714285613
992.506453,3286172.166,3285864.0459999996,0.04374999999999574,0.05528846153846045
992.50847,3286916.614,3286105.4353333334,0.03750000000000142,0.050520833333331794
992.510487,3287117.243,3286347.5138181816,0.02499999999999858,0.045454545454544165
992.512505,3287841.171,3286617.8825000003,0.012499999999995737,0.04062499999999858
992.514522,3288493.752,3286842.9967777785,0.006250000000001421,0.03541666666666524
992.516539,3288697.734,3287093.376625,0.0,0.030468749999998934
```

Zinput, 2025-02-11:
It it still failing to home and tap after running setup.

vvuk, 2025-02-12:
Hmm. Everything there looks fine. Are there any errors or any output after it fails to home? Check your klippy.log as well.

vvuk, 2025-02-27:
Apologies for not getting back to you -- I've been unable to figure out how to reproduce this. One thing to try -- set `home_trigger_safe_time_offset: 0.025` in your config and see if that makes a difference.  What printer is this on?

Zinput, 2025-03-10:
Sorry, I have put an unklicky on the printer by now because I needed a printer. It was on a Voron Switchwire running sensorless homing on the x axis, and tmc2209 stepper drivers. I think I will wait until the project matures more before I test it again or I'll try it if I can find a way to mount it to my Micron.

closed 2025-07-16 by vvuk