vvuk/eddy-ng #6: Cartographer over Canbus

WayneManion, 2025-02-13:
I could not get binaries built from this repo built to work on my Cartographer, which is connected via Canbus. 

It took much trial and error, but I was able to build, flash, and communicate with my Cartographer after I built the binary using these menuconfig options:

[image]

The key was adding "Set pin PA1 at microcontroller startup." I also had to add this to the menuconfig for Katapult. 

Note: if you want to build Katapult from source you will need to use Cartographer's Katapult fork. The required 24MHz crystal clock speed option is not present in mainline Katapult. 

I don't know yet if eddy-ng works on a Canbus Cartographer, but I was at least able to build the binary, flash it, and communicate with the MCU.

vvuk, 2025-02-15:
Yeah, apologies -- Cartographer needs to be much better documented. It's also a very tiny device (stm32f042) that requires some hacks. Specifically, there's hard-pwm support for the f042 that's not present in mainline Klipper that's required -- you can get them from [this commit](https://github.com/vvuk/klipper/commit/b17c55ed7c0ac5caaffd09b0c6c3596290082367) (the changes in the stm32 directory).

vvuk, 2026-11-01:
No precompiled firmware, but the Cartographer docs on `develop` (`docs/cartographer.md`) now have tested menuconfig settings for Katapult and Klipper (USB and CAN, including PA1 for CAN). The F042 hard-PWM patch is no longer needed: Klipper's `[static_pwm_clock]` provides the clock.

closed 2026-11-01 by vvuk