vvuk/eddy-ng #7: had everything working then suddenly starts failing taps

KD-4-Lyfe, 2025-02-15:
Just wanted to give some feedback.

took a while to get setup, and i had it going great!

However, I am having issues that seem like it could be heat related. Printing ABS so higher chamber temps, i noticed all the toolhead MCU temps getting up to 80-85C. I had a good 10 prints just testing out settings and all went perfect.

When i closed up the chamber, waited for a time the Tap function on start print stopped working.
PROBE_EDDY_NG_TEST_DRIVE_CURRENT all work and come back good, tried changing drive currents.
Just changed everything back to stock and allowed it to cooldown 

(drive current 15) and its all working again but (drive current 17) fails now.

All tests seem fine at 60C though i didn't try print. (only use abs)

added my klippy logs

4:37 PM
Internal error on command:"PRINT_START"
4:37 PM
Internal error on command:"PROBE_EDDY_NG_TAP"
4:37 PM
Klipper state: Shutdown
4:37 PM
Tap 5: failed (Error during homing probe: No samples received for time 0.000 (waited for 0.250))

[klippy (28).log](https://github.com/user-attachments/files/18809458/klippy.28.log)

KD-4-Lyfe, 2025-02-15:
Sorry, this is BTT Eddy Duo on canbus via the EBB36 board

vvuk, 2025-02-15:
Mm, there's a bug here,

vvuk, 2025-02-15:
Mm, there's a bug here -- the code tries to write the tap plot, but there aren't any samples, and that's why klippy crashes. I just fixed that.

However, that won't fix the underlying issue. To make sure I understand:

- dc 15 works fine at low chamber temperatures, but dc 17 fails
- dc 15 fails at higher chamber temps, but dc 17 works fine

is that correct?

KD-4-Lyfe, 2025-02-16:
dc 15 works at both temps, it was dc 17 that was working at higher chamber but not the lower.

I was just playing around trying different things with it.

pellcorp, 2025-02-22:
This reminds me so much of the issue I saw with classic btt eddy where if you calibrated it cold it worked hot but if you calibrated it hot then tried to use it colder than when it was calibrated it would fail with sensor errors during homing.  And the classic calibration gives you a higher driver current when it's hotter