vvuk/eddy-ng #8: SCREWS_TILT_CALCULATE issue

klich3, 2025-02-15:
I am not sure if it is a bug but when using the `eddy-ng` the `SCREWS_TILT_CALCULATE` command makes a kind of tap on the fourth point and then goes from 1 to 4 but without lowering the head, as if doing a read, which is normally not correct. Am I doing something wrong?

vvuk, 2025-02-16:
Hmm. Not sure what you mean. But it should move at the `horizontal_move_z` height and do a scan at that height. Can you describe the behaviour?

klich3, 2025-02-17:
I have an Ender converted to Switchwire with the CORE XZ, I level the table with M4 screws, previously the reading was done as if it were an endstop, now what it does is moderately correct that it passes through the point and makes the reading, but the reading is not correct, in all points it returns the same value.

klich3, 2025-02-21:
This is the three passes without adjustment of the table each time checking gives totally different values.

[image]

[image]

[image]

vvuk, 2025-02-23:
What is your horizontal_move_z set to in screws_tilt_adjust? Try running `SCREWS_TILT_CALCULATE horizontal_move_z=2` and see if that improves things.

klich3, 2025-02-23:
This is the configuration I have now with screws setting

'''text
[screws_tilt_adjust]
screw1: 60, 30
screw1_name: fl del. izq 1 
screw2: 230, 30
screw2_name: fr del. der 2
screw3: 230, 197
screw3_name: rr det. der 3
screw4: 60, 197
screw4_name: rl det. izq 4
horizontal_move_z: 2
speed: 300
screw_thread: CW-M3
'''

pudding1232, 2025-02-23:
try using your bed_mesh min and max x,y coordinates, you might be setting the eddy probe too close to the edge of the bed

vvuk, 2025-02-24:
Yeah, horizontal_move_z is already set in there. I just tested screws_tilt_adjust on my ender3 here and got valid values and adjustments.

However -- the values you show are not reading exactly the same value, so the sensor is working fine. And the values you get for each round are (mostly) within 0.010-0.020mm of eachother. There could be any number of very small issues in your kinematics that would cause this amount of difference.

One way to test would be to move the toolhead to some point, at Z=2. Run `PROBE_EDDY_NG_PROBE_STATIC`. If you run it more than once you should get values that are very close to eachother, if not identical.  Then move the toolhead by 50 x/y away diagonally, then move it back to the same spot (keeping Z=2 throughout). Then run `PROBE_EDDY_NG_PROBE_STATIC`.

For example, here's my ender3 v2 (completely stock), starting at 125,125,2 then moving to 175,175,2 and back to 125,125:

```
17:59:14  $ _CLIENT_LINEAR_MOVE Z=2 F=600 ABSOLUTE=1
17:59:18  $ peps
17:59:18  // Probed 2.007 (avg=2.006, 1.973 to 2.042, [0.015])
17:59:20  $ peps
17:59:20  // Probed 2.007 (avg=2.005, 1.973 to 2.033, [0.013])
17:59:21  $ peps
17:59:21  // Probed 2.007 (avg=2.008, 1.973 to 2.042, [0.013])
17:59:27  $ _CLIENT_LINEAR_MOVE X=175 Y=175 F=7800 ABSOLUTE=1
17:59:35  $ _CLIENT_LINEAR_MOVE X=125 Y=125 F=7800 ABSOLUTE=1
17:59:44  $ peps
17:59:44  // Probed 1.998 (avg=2.008, 1.989 to 2.024, [0.017])
17:59:46  $ peps
17:59:46  // Probed 1.998 (avg=1.997, 1.963 to 1.998, [0.006])
17:59:47  $ peps
17:59:47  // Probed 1.998 (avg=1.997, 1.963 to 2.024, [0.008])
```

If you get wildly different values when probing the exact same spot (without moving) that's an issue. But if you return back to the same position and probe you can expect a variance.

klich3, 2025-02-24:
Hi VVUK, thanks for the answer, this evening I check it, I am also checking the complete configuration, because when using the script of eddy's branch and now yours there are several points that rewrite gcode commands, it is possible that it can also come from there, anyway I will report.

klich3, 2025-02-24:
I am testing, and just what you say, I am doing the probe static from the same point and the image attached is fluctuating.
The previous method I guess what it would do is to make 2 or 3 readings and take an average. Obviously if it is one minute nothing happens since the range is 0 to 5 minutes of rotation is normal.

[image]

klich3, 2025-02-24:
I have checked the logic of Bigtech's Eddy code and I have found the solution, now the reading of the 4 dots does not jump as much as before. I have tested it several times and now it responds correctly.

It would be to check the rest of the things...
Regards

klich3, 2025-02-25:
For now I think the issue can be closed, more or less the issue has been found.

closed 2025-02-25 by klich3