vvuk/eddy-ng #10: Tap fails

cgarwood82, 2025-02-19:
When i attempt a tap, it fails and reports it's trying to go to a negative number:

```03:08:13  $ PROBE_EDDY_NG_TAP TARGET_Z=-0.150 THRESHOLD=250 SAMPLES=1
03:08:14  // Tap 1: failed (Error during homing move: Move out of range: 150.000 150.000 -0.150 [0.000])
03:08:15  // Tap 2: failed (Error during homing move: Move out of range: 150.000 150.000 -0.150 [0.000])
03:08:15  // Tap 3: failed (Error during homing move: Move out of range: 150.000 150.000 -0.150 [0.000])
03:08:16  // Tap 4: failed (Error during homing move: Move out of range: 150.000 150.000 -0.150 [0.000])
03:08:17  // Tap 5: failed (Error during homing move: Move out of range: 150.000 150.000 -0.150 [0.000])
03:08:17  !! Tap failed: 5 errors, last: Error during homing move: Move out of range: 150.000 150.000 -0.150 [0.000] at toolhead z=3.000
03:08:17  !! Tap failed```

Meshing works okay granted the bed is terrible, but I was going to start tramming but I can't get tap to work. Have I misconfigured something?

straga, 2025-02-19:
try: PROBE_EDDY_NG_TAP TARGET_Z=-0.250 THRESHOLD=250 SAMPLES=1
and show tap.html from /tmp/tap*

[image]

vvuk, 2025-02-19:
You'll need to set `position_min: -5` (or -2 or something) in your `stepper_z` configuration. Tap needs to be able to target below 0 to get an accurate read, because it needs to make contact with the build plate.. and depending on homing/thermal expansion/etc., true z=0 may be below the current homed 0 on the rail.

cgarwood82, 2025-02-20:
Thanks for the clarity. I've made the adjustments but tap still doesn't tap consistently. Sometimes it'll register a tap sometimes not. Something else noteworthy is I've not observed it making physical contact with my bed. Here is the output from an attempt with no success

tap 1

[image]

tap 2

[image]

tap 3

[image]

tap 4

[image]

tap 5

[image]

[tap-samples.csv](https://github.com/user-attachments/files/18880280/tap-samples.csv)

For verbosity, here are the relevant bits from my configuration

```
[mcu eddy]
# Or whatever is appropriate for CAN
serial: /dev/serial/by-id/usb-Klipper_rp2040_45474E621B06C63A-if00 
restart_method: command

[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu
sensor_mcu: eddy
min_temp: 10
max_temp: 100

[temperature_sensor btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26

# For all Eddy (USB/Duo/Coil)
[probe_eddy_ng btt_eddy]
sensor_type: ldc1612
# Eddy USB/Duo values below. For Eddy Coil, replace these two with mcu/bus for your setup.
i2c_mcu: eddy
i2c_bus: i2c0f
samples_per_second: 500 # 250 by default
x_offset: -17.25 # INSERT VALUE FOR YOUR PROBE POSITION
y_offset: -26.63 # INSERT VALUE FOR YOUR PROBE POSITION
tap_start_z: 3
tap_target_z: -0.250

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe_eddy_ng btt_eddy]
#*# calibration_16 = gASV5AoAAAAAAAB9lCiMAWaUXZQoR0FIQMxM6pbPR0FIQhi9lup+R0FIQ2UuQz4tR0FIRLGe75HdR0FIRf4Pm+WMR0FIR0qASDk7R0FISJbw9IzqR0FISeNhoOCZR0FISy/STTRJR0FITHxC+Yf4R0FITcizpdunR0FITxUkUi9WR0FIUGGU/oMFR0FIUa4Fqta1R0FIUvp2VypkR0FIVEbnA34TR0FIVZNXr9HCR0FIVt/IXCVxR0FIWCw5CHkhR0FIWXiptMzQR0FIWsUaYSB/R0FIXBGLDXQuR0FIXV37ucfdR0FIXqpsZhuMR0FIX/bdEm88R0FIYUNNvsLrR0FIYo++axaaR0FIY9wvF2pJR0FIZSifw734R0FIZnUQcBGoR0FIZ8GBHGVXR0FIaQ3xyLkGR0FIalpidQy1R0FIa6bTIWBkR0FIbPNDzbQUR0FIbj+0egfDR0FIb4wlJltyR0FIcNiV0q8hR0FIciUGfwLQR0FIc3F3K1aAR0FIdL3n16ovR0FIdgpYg/3eR0FId1bJMFGNR0FIeKM53KU8R0FIee+qiPjsR0FIezwbNUybR0FIfIiL4aBKR0FIfdT8jfP5R0FIfyFtOkeoR0FIgG3d5ptYR0FIgbpOku8HR0FIgwa/P0K2R0FIhFMv65ZlR0FIhZ+gl+oUR0FIhuwRRD3ER0FIiDiB8JFzR0FIiYTynOUiR0FIitFjSTjRR0FIjB3T9YyAR0FIjWpEoeAvR0FIjra1TjPfR0FIkAMl+oeOR0FIkU+Wpts9R0FIkpwHUy7sR0FIk+h3/4KbR0FIlTToq9ZLR0FIloFZWCn6R0FIl83KBH2pR0FImRo6sNFYR0FImmarXSUHR0FIm7McCXi3R0FInP+MtcxmR0FInkv9YiAVR0FIn5huDnPER0FIoOTeusdzR0FIojFPZxsjR0FIo33AE27SR0FIpMowv8KBR0FIphahbBYwR0FIp2MSGGnfR0FIqK+CxL2PR0FIqfvzcRE+R0FIq0hkHWTtR0FIrJTUybicR0FIreFFdgxLR0FIry22Il/7R0FIsHomzrOqR0FIscaXewdZR0FIsxMIJ1sIR0FItF940663R0FItavpgAJnR0FItvhaLFYWR0FIuETK2KnFR0FIuZE7hP10R0FIut2sMVEjR0FIvCoc3aTSR0FIvXaNifiCR0FIvsL+NkwxR0FIwA9u4p/gR0FIwVvfjvOPR0FIwqhQO0c+R0FIw/TA55ruR0FIxUExk+6dR0FIxo2iQEJMR0FIx9oS7JX7R0FIySaDmOmqR0FIynL0RT1aR0FIy79k8ZEJR0FIzQvVneS4R0FIzlhGSjhnR0FIz6S29owWR0FI0PEnot/GR0FI0j2YTzN1R0FI04oI+4ckR0FI1NZ5p9rTR0FI1iLqVC6CR0FI129bAIIyR0FI2LvLrNXhR0FI2gg8WSmQR0FI21StBX0/R0FI3KEdsdDuR0FI3e2OXiSeR0FI3zn/CnhNR0FI4IZvtsv8R0FI4dLgYx+rR0FI4x9RD3NaR0FI5GvBu8cKR0FI5bgyaBq5R0FI5wSjFG5oR0FI6FETwMIXR0FI6Z2EbRXGR0FI6un1GWl2R0FI7DZlxb0lR0FI7YLWchDUR0FI7s9HHmSDR0FI8Bu3yrgyR0FI8WgodwvhR0FI8rSZI1+RR0FI9AEJz7NAR0FI9U16fAbvR0FI9pnrKFqeR0FI9+Zb1K5NR0FI+TLMgQH9R0FI+n89LVWsR0FI+8ut2albR0FI/Rgehf0KR0FI/mSPMlC5R0FI/7D/3qRpR0FJAP1wivgYR0FJAknhN0vHZYwBaJRdlChHQC3+PXCj1wpHQCou1apOijRHQCcNREXgo3BHQCT1hXFmiIpHQCNup4h7HshHQCIJs7GbMNNHQCD38uhrPQhHQCAIStECRTZHQB5yGVox9G9HQBzab3VtWHJHQBudgfeFHjdHQBpcJlzjJ+BHQBlhFwatufZHQBhbR6jGr+JHQBeJFt1TP7hHQBanHrwoPtdHQBXk/mnUAqZHQBU0Qn9hPORHQBSEmV7z1l1HQBPupesnFZ5HQBNVwaT5MI9HQBK5uheOB4ZHQBI4YNsjBt1HQBG11bk/dPtHQBE/6ECHq5xHQBDUekWldE5HQBBnrRIkNxNHQA/4SMEJPc9HQA818osqWCVHQA55gN/CXElHQA29XLffxLVHQA0NBX67hcZHQAxkIZ8GhVZHQAvG7emjk3FHQAsnTXpENVZHQAqGKZ0I+fdHQAnuJa8B5/FHQAljNHNE3NlHQAjcCL8YMFhHQAhTsjUxllJHQAfStU2/lXVHQAdUJ1invcFHQAbXZgHOy9hHQAZanRc6gDNHQAXiX6GK+mRHQAVyUXkVS7lHQAUJa/nGIjxHQASghgka/+5HQAQ0IurCKVFHQAPRVJMcDaxHQANq3YP0fGJHQAMLe/kHl5pHQAKpkBfKnGFHQAJMUi73YypHQAHxBy9PVLpHQAGVqRUy9exHQAFA5HekLONHQADqes7CulZHQACVRWXK0w5HQABCqC4HI1FHP//jx2jw989HP/89sljBAHxHP/6bD8DyKH5HP/4OdxnUAbJHP/1/y14dbuhHP/z1iFvxxbpHP/xrEIor5nZHP/varimaaHpHP/tJ3bOGtINHP/rDXFnlmoZHP/o33m2KxZ1HP/m5y07E4VFHP/k41ZtYKyVHP/i81ZGSQkxHP/g81BgRxH5HP/fAgsozFPBHP/dERLGDTKhHP/bOBLmaGj5HP/ZVXc8ohWZHP/Xa/ltYkAFHP/Voy+XgDQtHP/T31bHDVORHP/SI2pfpGONHP/QZLnVAhC1HP/OuGzmn8h1HP/NI54cGGn9HP/Lf4/Xc/gFHP/J3UE0fbllHP/IXZ9Q09uJHP/GuAG73FcFHP/FLIVBGn7RHP/DkqNFxA51HP/CEUXang8NHP/AgJy8sNqlHP+951pbS0otHP+7CCYnl+HpHP+4JwHskzBxHP+1LmNICoSVHP+yUVTJoFR1HP+viPzijkwVHP+suLArrm4hHP+qFJVUFL5dHP+nWtInAWmtHP+ksZrUdCjtHP+h53k/inoVHP+fTJZ0WD9FHP+cusEut3mtHP+aFwK3nXztHP+XheF0vZHFHP+VDQ1GBnRNHP+ShqYjmhXtHP+QOobkGBmZHP+N8vJ3trJ9HP+LjUuGxLLZHP+JMw7Peq9tHP+G3xjNvKd5HP+EiLC26djFHP+CCOAxXRZ9HP9/i+HzPHjVHP96yDxZTw+dHP92E7OaXLJBHP9xcML2TXZtHP9s4nlDq+mdHP9oX9FgHN8lHP9kBL/pnST9HP9fu2gg2MjlHP9bePZb0thdHP9XNNMiAyAhHP9S//O/ble1HP9O1919AZPhHP9KvyGOOeC1HP9GnnatxxsxHP9Ch/Zy8JshHP88hpAe7TSJHP80BzEJx94pHP8rqFnjAb2JHP8jf4aqYURVHP8bM52w7tsdHP8TNa2E0o85HP8LhX42mMbdHP8Dm4519HrhHP73XCSRe0IRHP7n8SADeFHxHP7ZYjgOUjMdHP7K2jgHdaQRHP64QtO+HpbJHP6bW20hYhMxHP57ZuBMRBxhHP5E+DOesDOBHP264UeuFHgBljAdoX3JhbmdllEc/brhR64UeAEdALf49cKPXCoaUjAdmX3JhbmdllEdBSEDMTOqWz0dBSQJJ4TdLx4aUjAJkY5RLEHUu
#*# calibrated_drive_currents = 16
#*# calibration_version = 3
#*# saved_reg_drive_current = 16
#*# saved_tap_drive_current = 16
```

cgarwood82, 2025-02-20:
Here is a single successful tap with all of the relevant details:
```
22:22:55  $ PROBE_EDDY_NG_TAP TARGET_Z=-0.250 THRESHOLD=250 SAMPLES=1
22:22:57  // Tap 1: failed (Error during homing probe: Sensor error (ldc status: UNREADCONV1 ERR_ALE))
22:22:59  // Tap 2: z=0.295
22:23:01  // Probe computed z offset 0.295 (tap at z=0.295, stddev 0.000), sensor offset 0.044 at z=2.000
```

Tap 1

[image]

Tap 2

[image]

[tap-samples.csv](https://github.com/user-attachments/files/18880370/tap-samples.csv)

Looking at this graph, it appears that there are registered taps with higher frequencys, but again, I don't actually observe physical contact between the nozzle and the bed.

cgarwood82, 2025-02-23:
Any ideas on what else I can try before I remove the probe?

Sodakky, 2025-02-23:
> Any ideas on what else I can try before I remove the probe? 

Try running PROBE_EDDY_NG_CALIBRATE DRIVE_CURRENT=17,

And then in your eddy config file add this line  : tap_drive_current: 17

What I'm guessing what is happening is a drive current of 16 (that you got from the auto setup) is right on the threshold of working/not working for tap, so when your ran the setup it just happened to fit the spec at that time when it tried it.

vvuk, 2025-02-24:
> Looking at this graph, it appears that there are registered taps with higher frequencys, but again, I don't actually observe physical contact between the nozzle and the bed.

The spikes on the left side of the graph are ignored (even though they show up on the graph). The one that seems to trigger doesn't look like a tap for sure. There are a couple things you can try. One is what @Sodakky suggested -- try using a higher drive current for tap after calibrating.

Another is to increase the threshold. You can also do something like this:

```
..._TAP THRESHOLD=2000 TARGET_Z=-0.200 SAMPLES=1
```

which won't successfully complete (or shouldn't) because of the high threshold, but _should_ make contact with the build plate. Then you can look at the tap graph and see if there's a better threshold value. It looks like what's triggered as a tap there is the sensor registering just barely a slowdown which could be for a number of reasons (slightly too low sensor which causes odd effects when the toolhead gets very close to the build plate; tiny bit of roughness on z axis there if you've got leadscrews or something; gremlins). Post that graph if you don't mind as well.

A third thing would be to try raising the sensor slightly.

Also, if you update to the latest, the graphs will start at a more useful X range making it easier to see what's happening (you can also drag zoom manually), and there's a new config `max_errors` that you can set to say 3 or 4, which will allow that many invalid samples in a row (which may resolve the first issue, along with higher drive current). The default is 0, meaning any error aborts.

My guess is a higher threshold should sort this out.

cgarwood82, 2025-02-25:
Looks like I had it mounted too low. I printed a 2.75 shim to put the eddy on but must have been off a little I suppose. Regardless, this flashing with the latest firmware and adjusting my current seems to have got me back into business!

``` 21:39:58  $ PROBE_EDDY_NG_TAP
21:40:00  // Tap 1: z=0.853
21:40:03  // Tap 2: z=0.855
21:40:05  // Tap 3: z=0.895
21:40:07  // Probe computed tap Z at 0.868 (tap at z=0.868, stddev 0.019), sensor offset 0.019 at z=2.000```

Eddy calibration:
![Image](https://github.com/user-attachments/assets/24d3dcd4-d076-4736-b1f4-46103e3a79b6)

Tap 1

![Image](https://github.com/user-attachments/assets/7d3a5fd8-81e2-44c6-8c38-75eb8d745b72)

Tap 2

![Image](https://github.com/user-attachments/assets/edf16a9c-840e-4af4-ab5a-80a936b4e119)

Tap 3

![Image](https://github.com/user-attachments/assets/7bcec0d4-9cd5-4517-bd0e-c9c9504686f1)

Is there much benefit to calibrate again without specifying a hard coded current? Thanks

cgarwood82, 2025-02-25:
I think every time I've run tap, it's always had it around .8 or .9. When I run a print, it prints perfect first layers from the limited testing I've done tonight...BUT, I cannot got to z0 because it's about 1mm into my bed if I was to full send it 😓. Kicking of a print though, the offset is used and we are off running perfectly. No z adjustment has been needed. 

I'll say though, this is a nearly completely new printer from various parts and I'm still getting my klipper legs. What else can I check on my end?

vvuk, 2025-02-25:
> BUT, I cannot got to z0 because it's about 1mm into my bed if I was to full send it

Yeah this makes sense! If tap thinks the bed is at Z=0.8.. then Z=0 is definitely very much into the bed :) Couple things to try.. first, do a full home (G28). Move the toolhead to Z=3.0. Run `PROBE_EDDY_NG_PROBE_STATIC`. The value you get back should be near 3.0. (Can also do this at 2.0.) If it's not, one of two things is happening:

1. Calibration is badly off. Move the toolhead back to Z=3 and then measure with a ruler the distance between nozzle and bed. It should be roughly 3.0ish. If it's off, then recalibrate.
2. Something in your homing macro/setup is homing to the wrong value. Post your `homing_override` or `G28` macro override here and I can take a look, but generally for homing Z you want a `homing_override`, with the Z sequence being something like:

```
G28 Z    # do a coarse triggered Z home
G90    # absolute positioning
G0 Z2 F1000    # move toolhead to z=2mm
PROBE_EDDY_NG_PROBE_STATIC HOME_Z=1    # home precisely using the probe
G0 Z5 F1000    # z-hop up a bit to finish up
```

closed 2025-03-09 by cgarwood82