vvuk/eddy-ng #11: How to deal with multiple sheets?

DomiStyle, 2025-02-19:
First off, thanks for making this. It works really well once calibrated with the default Voron 2.4r2 X carriage in the lowest possible position.

One issue I am having is with switching sheets. I have a PEI, Textured and Satin sheet and I was wondering what the correct procedure is to deal with all of them?

If I calibrate with the PEI sheet everything works fine, tap only needs a `Z_TARGET` of `-0.200` and first layer looks pretty good.

Switching to the Satin sheet requires me to set a `Z_TARGET` of `-0.600`. Which I guess makes sense since the Satin sheet has no plastic above the metal sheet.

It will work perfectly fine on the Satin sheet with such a large `Z_TARGET` but I'm just not sure if it is smart to set such a large value? Should I recalibrate on the Satin sheet? Should I keep calibration for every sheet? Are there any recommendations for this scenario?

vvuk, 2025-02-19:
Should be `TARGET_Z` (vs `Z_TARGET`) -- I had an inconsistency in the commands where a few took `Z_TARGET`, but tap always took `TARGET_Z`; if you were setting Z_TARGET for tap it wasn't doing anything :)

But in general a low TARGET_Z is not an issue -- if tap is successful, the toolhead will never reach that point since it will stop moving as soon as tap is detected. In other words, `TARGET_Z=-5.000` should behave exactly the same as `TARGET_Z=-0.200` if tap is detected at something like -0.050. So there isn't much downside to using a lower target_z; it's more of a safety check than anything, because if for whatever reason tap isn't detected, the Z steppers will actually try to move all the way down to it. But -0.600 should still be fine.. even in the case of failure, it'll be a good bit of pressure on the build plate especially on the PEI sheet, but shouldn't damage anything (not a guarantee! :) .. and again if things are working as they should, it'll never actually reach that point.

Another thing you may need to change per build sheet is any `tap_adjust_z` value (or you may not -- all depends on where the flex is that you're adjusting for); if you do, there is a `PROBE_EDDY_NG_SET_TAP_ADJUST VALUE=...` command that you could insert in your start print macro based on the sheet coming from the slicer.

DomiStyle, 2025-02-20:
> Should be TARGET_Z (vs Z_TARGET) -- I had an inconsistency in the commands where a few took Z_TARGET, but tap always took TARGET_Z; if you were setting Z_TARGET for tap it wasn't doing anything :)

My fault, I use `TARGET_Z` so everything should be working. :)

> But -0.600 should still be fine.. even in the case of failure, it'll be a good bit of pressure on the build plate especially on the PEI sheet, but shouldn't damage anything (not a guarantee! :) .. and again if things are working as they should, it'll never actually reach that point.

Alright, I will leave it at -0.600 for now then and do some prints. My PEI sheet is already messed up from my old Voron Tap with 4 magnets, so not much getting destroyed there.

> Another thing you may need to change per build sheet is any tap_adjust_z value (or you may not -- all depends on where the flex is that you're adjusting for); if you do, there is a [...]

Will add a macro in my start gcode which takes care of the sheet selection. So far works pretty well though with the same `tap_adjust_z` for all sheets.

Thanks!

closed 2025-02-20 by DomiStyle