vvuk/eddy-ng #17: unable to z home from certain height

ttpimpin, 2025-02-24:
hey i've searched to see if it's been posted before (please forgive me if i didn't look hard enough) but i have the mellow fly sensor. i have tap set up, and i'm able to scan the bed. only issue is i can only home the z if it's really close to the bed. if i finish a print, and i try to home the printer, x/y homes and when it comes to z, "Error during homing z: Sensor error (ldc status: UNREADCONV1 DRDY)". so i have to manually raise the bed to around 5-10mm close to the probe. again, sorry if the answer was stated before.

vvuk, 2025-02-27:
This could be similar to #3 that I've had a hard time reproducing. One thing to try -- set `max_errors: 3` in the config, see if that helps. (This allows 3 errors in a row to not abort homing or other operations, in case an error is spurious.)

Otherwise, can you move the toolhead to about 15mm off the bed and run `PROBE_EDDY_NG_STATUS` and `PROBE_EDDY_NG_PROBE_STATIC` and paste the results? Then do the same at 50mm and 100mm. (Doesn't have to be exact, just "far away"... also ignore the height readings from those, they'll be incorrect :)

vvuk, 2025-02-27:
Oh also -- what printer & what build plate?

cattivik66, 2025-06-04:
I have the same issue.
Printer: Qidi X-Plus 3

It is a printer where the bed moves up and down, and doesn't have any Z endstops.

When you turn on the printer after some time, the bed is at the bottom of the printer, and Eddy distance is about ~32cm.

I think the SAVE Z HOME should have a logic like that (sorry, I am not a G-Code expert, so I write some pseudo-code):

PROBE_EDDY_NG_PROBE_STATIC ; check if the probe can read the current height
WHILE (probed position > 5mm or probed position is undefined)
   FORCE_MOVE STEPPER=stepper_z DISTANCE=-5 VELOCITY=10 ; move up the bed by 5mm
   PROBE_EDDY_NG_PROBE_STATIC
END

deece, 2025-09-05:
I see similar on a Voron V0.2 with a Fly SHT36V3 + eddy sensor. Similar setup to the above, Z movement is on the bed. Magnetic build plate base, spring steel with holographic layer. I see similar results with the Siboor spring steel PEI sheet included in the Siboor Voron 0.2 kit.

If the bed is too far, I get:
```
Error during homing z: No samples received for time 0.000 (waited for 0.250)
PROBE_EDDY_NG_STATUS
Last coil value: 0.00 (-infmm) raw: 0x137bc3c3 ERROR: 0b1 status: 0x408 UNREADCONV1 ERR_AHE
```

Setting max_errors: 3 had no effect.

Setting home_trigger_safe_time_offset: 0.025 gives:
```
Option 'home_trigger_safe_time_offset' is not valid in section 'probe_eddy_ng fly_eddy'
```

Setting the same manually in probe_eddy_ng.py has no effect.

**0mm away**
```
Last coil value: 4492901.64 (0.264mm) raw: 0x3982571 status: 0x48 UNREADCONV1 DRDY
Probed 0.258 (avg=0.258, 0.258 to 0.259, [0.000])
```

**2mm away**
```
Last coil value: 4420781.96 (2.006mm) raw: 0x389604b status: 0x48 UNREADCONV1 DRDY
Probed 2.006 (avg=2.006, 2.005 to 2.006, [0.000])
```

**5mm away**
```
Last coil value: 4367034.14 (4.973mm) raw: 0x37e5e5c status: 0x408 UNREADCONV1 ERR_AHE
Probed 4.973 (avg=4.972, 4.970 to 4.973, [0.002])
```

**6mm away**
```
Last coil value: 4358518.57 (5.908mm) raw: 0x37c9fe6 status: 0x408 UNREADCONV1 ERR_AHE
Probed 5.933 (avg=5.933, 5.932 to 5.935, [0.001])
```

**6.5mm away**
```
Last coil value: 4355250.45 (6.330mm) raw: 0x37bf48e status: 0x48 UNREADCONV1 DRDY
Probed 2.083 (avg=2.040, 1.959 to 2.083, [0.073])
```
**6.75 mm away**
```
Last coil value: 0.00 (-infmm) raw: 0x137ba5c0 ERROR: 0b1 status: 0x48 UNREADCONV1 DRDY
Probed -12.217 (avg=-12.424, -12.666 to -12.217, [0.304])
```

**7mm away**
```
Last coil value: 0.00 (-infmm) raw: 0x137b5c8f ERROR: 0b1 status: 0x48 UNREADCONV1 DRDY
Probed -53.527 (avg=-53.527, -53.527 to -53.527, [0.000])
```

**8mm away**
``` 
Last coil value: 0.00 (-infmm) raw: 0x137a6923 ERROR: 0b1 status: 0x48 UNREADCONV1 DRDY
No samples received for time 632.798 (waited for 0.351)
```

**15mm away**
```
Last coil value: 0.00 (-infmm) raw: 0x1377afb1 ERROR: 0b1 status: 0x408 UNREADCONV1 ERR_AHE
No samples received for time 917.944 (waited for 0.351)
```

**50mm away**
```
Last coil value: 0.00 (-infmm) raw: 0x1376ac4b ERROR: 0b1 status: 0x408 UNREADCONV1 ERR_AHE
No samples received for time 962.640 (waited for 0.351)
```

**100mm away**
```
Last coil value: 0.00 (-infmm) raw: 0x1376a582 ERROR: 0b1 status: 0x408 UNREADCONV1 ERR_AHE
No samples received for time 1026.477 (waited for 0.351)
```

deece, 2025-09-05:
I'm also running a BTT Eddy on my Voron 2.4, and when I move that away from the bed (say, 100mm), I see:
```
Last coil value: 3206922.01 (9.048mm) raw: 0x446a11f status: 0x48 UNREADCONV1 DRDY
Probed 10.858 (avg=10.827, 10.628 to 11.119, [0.130])
```

(which is obviously incorrect, but sufficient to keep homing from breaking)

deece, 2025-09-06:
My Voron 2.4 does show similar symptoms intermittently.

@vvuk I wonder if a quick and easy workaround for now is, when homing, return a height greater than the range of the sensor, if the sensor does not return a reading?

vvuk, 2026-11-01:
Two things have changed here. Homing no longer aborts when the sensor reports a too-strong signal (ERR_AHE), which is what it does far from the bed with too high a drive current; it assumes the toolhead is high and continues. And the recent "homes in mid-air from 5cm" reports are the new-Klipper problem fixed on `main` (see #146). If you still see sensor errors when homing from high up, try a drive current one lower than what setup picked.