vvuk/eddy-ng #23: Internal error on command:"PROBE_EDDY_NG_TAP"

wysockiw, 2025-02-27:
[klippy.log](https://github.com/user-attachments/files/19001860/klippy.log)

Getting Internal error on command:"PROBE_EDDY_NG_TAP" error during initial TAP calibration - all the previous steps finished successfully

vvuk, 2025-02-27:
Hmm, looks like you've got a broken `scipy` install:

```
File "/home/pi/klipper/klippy/extras/probe_eddy_ng.py", line 2281, in _write_tap_plot
    butter_s_t, butter_s_v = self._compute_butter_tap(
  File "/home/pi/klipper/klippy/extras/probe_eddy_ng.py", line 1929, in _compute_butter_tap
    sos = scipy.signal.butter(
  File "/home/pi/klippy-env/lib/python3.9/site-packages/scipy/__init__.py", line 134, in __getattr__
    return _importlib.import_module(f'scipy.{name}')
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
...
  File "/home/pi/klippy-env/lib/python3.9/site-packages/scipy/linalg/_misc.py", line 3, in <module>
    from .blas import get_blas_funcs
  File "/home/pi/klippy-env/lib/python3.9/site-packages/scipy/linalg/blas.py", line 213, in <module>
    from scipy.linalg import _fblas
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory
```

In this case it's only used for the tap plot, so if it's difficult to fix the install, you could just uninstall it (`~/klippy-env/bin/pip3 uninstall scipy`) or disable writing tap plots with setting `write_tap_plot: false` in the config.

closed 2025-02-27 by vvuk

wysockiw, 2025-02-28:
Hmm I'm afraid it is something else - uninstalled scipy but the same happening :-(  still getting Internal error on command:"PROBE_EDDY_NG_TAP" error 

[klippy (2).log](https://github.com/user-attachments/files/19019570/klippy.2.log)