vvuk/eddy-ng #24: Internal error on command:"PROBE_EDDY_NG_TAP"

wysockiw, 2025-02-28:
Hmm I'm afraid it is something else - uninstalled scipy but the same happening :-( still getting Internal error on command:"PROBE_EDDY_NG_TAP" error

[klippy (2).log](https://github.com/user-attachments/files/19019585/klippy.2.log)

vvuk, 2025-02-28:
No, it's the same problem. You can search the log for "Traceback" (ignore the ones about unable to obtain bulk status response).

wysockiw, 2025-02-28:
Thank you for reply and checking this for me - this time I uninstalled scipy and plotly - rebooted Raspi , deleted all the logs to get clean not confusing logs and now the error goes away ... now getting different errors but with these I will go and try troubleshooting issue and will go back if I will not be able figure that out on my own 

[image]

closed 2025-03-02 by wysockiw