vvuk/eddy-ng #26: Error: Heater extruder not heating at expected rate and Error: MCU 'mcu' shutdown: Timer too close

lemonhole, 2025-03-04:
Hi. 
I'm experiencing frequent errors after bed scanning on my Elegoo Neptune 3 Plus with a BTT Pi 1.2 and Eddy. 
This issue started after installing eddy-ng, and it didn't occur before. 
I'm attaching my printer.cfg (as .txt because can't add .cfg) and start G-code
[printer (5).txt](https://github.com/user-attachments/files/19072439/printer.5.txt)

G90 ; use absolute coordinates
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

M106 S102 ; fan at 40%
M140 S{first_layer_bed_temperature[0]} ; start to heat bed
M104 S160 ; set temporary nozzle temp to prevent oozing
G28 X Y
M190 S[first_layer_bed_temperature] ; wait for bed temp to stabilize

;M106 S179 ; fan at 70%
G28 Z
PROBE_EDDY_NG_TAP
BED_MESH_CALIBRATE METHOD=rapid_scan

G1 X4 Y5 F3000 ; move to left corner
G1 Z0.28 ; move nozzle down to 0.28
M106 S0 ; fan at 0%

M104 S[first_layer_temperature] ; set final nozzle temp
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize

RESPOND PREFIX=tgnotify MSG="purge line"
G1 Z0.3 F240 ; move nozzle up to 0.3
G92 E0 ;Reset Extruder

G1 Y100 E10 F1500 ; prime the nozzle
G1 X4.3 F5000
G92 E0
G1 Y3 E10 F1200 ; prime the nozzle
G1 X4.6 F5000
G92 E0
G1 Y95 E10 F1200 ; prime the nozzle

G92 E0 ;Reset Extruder

lemonhole, 2025-03-04:
The "MCU 'mcu' shutdown: Timer too close" error only happened once, and I haven't been able to reproduce it, so I removed it from the title

vvuk, 2025-03-04:
But the heater issue is still happening? There shouldn't be anything being written to the SD card during bed mesh, but there may be more data streaming in than before. Can you upload your klippy log when it happened (either/both)? Also the printer.cfg you attached seems to be from before you moved to eddy-ng (it doesn't reference it, references old eddy code)

lemonhole, 2025-03-04:
@vvuk 
My apologies, that was backup of printer.cfg. I've now attached the correct one. 
[printer.txt](https://github.com/user-attachments/files/19074895/printer.txt)

Reproduced the error. You can find it by searching for "expected":
[klippy.log](https://github.com/user-attachments/files/19075048/klippy.log)

vvuk, 2025-04-20:
This _should_ be fixed on latest `main`; there were a lot of performance improvements in specifically the step that's slow here. Please reopen this if you still see this issue after updating!

closed 2025-04-20 by vvuk