vvuk/eddy-ng #27: Flashing STM32F072 errors

ItsReckliss, 2025-03-09:
Hi, I was able to flash the modified klipper firmware (with the ng drivers) onto my eddy, octopus pro and EBB36, but when I am trying to build the firmware to flash it to my Fly DP5 (used for custom afc) I get this error:

`biqu@BTT-CB1:~/klipper$ make clean
biqu@BTT-CB1:~/klipper$ make menuconfig
  Creating symbolic link out/board
Loaded configuration '/home/biqu/klipper/.config'
Configuration saved to '/home/biqu/klipper/.config'
Configuration saved to '/home/biqu/klipper/.config'
Configuration saved to '/home/biqu/klipper/.config'
biqu@BTT-CB1:~/klipper$ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_hx71x.o
  Compiling out/src/sensor_ads1220.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_ldc1612_ng.o
src/sensor_ldc1612_ng.c: In function 'config_ldc1612_ng':
src/sensor_ldc1612_ng.c:342:9: warning: implicit declaration of function 'gpio_pwm_setup_with_max'; did you mean 'gpio_pwm_setup'? [-Wimplicit-function-declaration]
         gpio_pwm_setup_with_max(GPIO('B', 4), 1, 1, 2);
         ^~~~~~~~~~~~~~~~~~~~~~~
         gpio_pwm_setup
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f0/system_stm32f0xx.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/stm32/stm32f0_timer.o
  Compiling out/src/stm32/stm32f0.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/stm32f0_adc.o
  Compiling out/src/stm32/stm32f0_i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/usbfs.o
  Compiling out/src/stm32/chipid.o
  Compiling out/src/generic/usb_cdc.o
  Compiling out/src/stm32/hard_pwm.o
  Building out/compile_time_request.o
Version: v0.12.0-410-gcf3b0475d-dirty-20250309_230541-BTT-CB1
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: out/src/sensor_ldc1612_ng.o: in function `config_ldc1612_ng':
/home/biqu/klipper/src/sensor_ldc1612_ng.c:342: undefined reference to `gpio_pwm_setup_with_max'
collect2: error: ld returned 1 exit status
make: *** [Makefile:73: out/klipper.elf] Error 1
biqu@BTT-CB1:~/klipper$ ls ~/klipper/out/klipper.bin
ls: cannot access '/home/biqu/klipper/out/klipper.bin': No such file or directory`

vvuk, 2025-03-10:
Ah, so the problem is that the eddy-ng code thinks it's being compiled for Cartographer because it's that same chip, and the pwm code for cartographer never made it into Klipper. Hrm! Easiest thing to get you unblocked... You don't actually have to flash anything else, just the eddy needs the new firmware (unless your other firmware is too old for other reasons). You've got the eddy flashed, so leave that alone. Edit `src/Makefile` and delete `sensor_ldc1612_ng.c` from where it shows up. Leave everything else on that line alone. Then build and flash your DP5 as normal.

vvuk, 2025-04-20:
This should be fixed (from a while back) -- i've disabled the by-default support for cartographer, so stm32f0 compilation should work fine.

closed 2025-04-20 by vvuk