vvuk/eddy-ng #28: Tap fails after idle for a while

Coare1, 2025-03-10:
I added PROBE_EDDY_NG_TAP to my print start gcode, but whenever I leave the printer idle for a while and come back to it tap fails the first time, then works the second time.
I'm running an Eddy Duo on the Sovol SV08 with the low Rappetor mount.

Here is the console output (newest to oldest):

```
18:04:54 - Probe computed tap Z at 0.034 (tap at z=0.034, stddev 0.006), sensor offset 0.051 at z=2.000
18:04:52 - Tap 5: z=0.026
18:04:50 - Tap 4: z=0.138
18:04:47 - Tap 3: z=0.170
18:04:44 - Tap 2: z=0.039
18:04:41 - Tap 1: z=0.037
18:04:38 - PROBE_EDDY_NG_TAP
18:04:36 - Result is z=1.957225
18:04:11 - Homing Z
18:04:11 - Homing X
18:04:11 - Homing Y
18:04:10 - G28
18:03:33 - Cancel Print Success!
18:03:33 - Tap failed
18:03:33 - Tap failed: 10 errors, last: Error during homing probe: Sensor error at toolhead z=3.001
18:03:33 - Tap 10: failed (Error during homing probe: Sensor error)
18:03:32 - Tap 9: failed (Error during homing probe: Sensor error (ldc status: UNREADCONV1 DRDY))
18:03:32 - Tap 8: failed (Error during homing probe: Sensor error)
18:03:31 - Tap 7: failed (Error during homing probe: Sensor error)
18:03:30 - Tap 6: failed (Error during homing probe: Sensor error)
18:03:30 - Tap 5: failed (Error during homing probe: Sensor error (ldc status: UNREADCONV1 ERR_AHE))
18:03:29 - Tap 4: failed (Error during homing probe: Sensor error (ldc status: UNREADCONV1 ERR_AHE))
18:03:28 - Tap 3: failed (Error during homing probe: Sensor error (ldc status: UNREADCONV1 ERR_AHE))
18:03:28 - Tap 2: failed (Error during homing probe: Sensor error (ldc status: UNREADCONV1 ERR_AHE))
18:03:27 - Tap 1: failed (Error during homing probe: Sensor error (ldc status: UNREADCONV1 DRDY))
```

vvuk, 2025-03-10:
Hmm -- so this was leaving the printer idle, enough for it to time out steppers (so printer becomes unhomed), and then doing a tap? Or something else?

I noticed that you homed in between, was the toolhead in a different location before the home?

vvuk, 2025-03-10:
Trying to tap after the steppers time out (so unhomed) correctly gives an error that Z must be homed. So there must be something else going on here. Anything else different between the frist and the second?

Coare1, 2025-03-10:
Yes, leaving it idle until the printer becomes unhomed, I can tell because the fans turn off. I home, clean the nozzle, and do a QGL before the tap though.

Here is the log before the tap:
```
18:03:24 - Retries: 2/5 Probed points range: 0.020489 tolerance: 0.050000
18:03:24 - Making the following Z adjustments:
stepper_z = -0.016133
stepper_z1 = 0.018488
stepper_z2 = -0.002081
stepper_z3 = -0.000273
18:03:24 - Average: 8.210827
18:03:23 - Actuator Positions:
z: 8.226960 z1: 8.192339 z2: 8.212908 z3: 8.211100
18:03:23 - Gantry-relative probe points:
0: 8.222178 1: 8.201689 2: 8.210311 3: 8.213478
18:03:18 - Retries: 1/5 Probed points range: 0.063088 tolerance: 0.050000
18:03:18 - Making the following Z adjustments:
stepper_z = -0.040412
stepper_z1 = 0.066554
stepper_z2 = -0.016741
stepper_z3 = -0.009400
18:03:18 - Average: 8.212576
18:03:18 - Actuator Positions:
z: 8.252988 z1: 8.146022 z2: 8.229317 z3: 8.221976
18:03:17 - Gantry-relative probe points:
0: 8.241299 1: 8.178211 2: 8.217633 3: 8.226367
18:03:12 - Retries: 0/5 Probed points range: 0.247082 tolerance: 0.050000
18:03:12 - Making the following Z adjustments:
stepper_z = -0.123125
stepper_z1 = 0.281445
stepper_z2 = -0.061208
stepper_z3 = -0.097112
18:03:12 - Average: 8.226559
18:03:12 - Actuator Positions:
z: 8.349684 z1: 7.945114 z2: 8.287768 z3: 8.323671
18:03:12 - Gantry-relative probe points:
0: 8.320216 1: 8.073789 2: 8.249748 3: 8.320872
18:03:04 - Retries: 1/5 Probed points range: 0.322829 tolerance: 1.000000
18:03:04 - Making the following Z adjustments:
stepper_z = -0.290128
stepper_z1 = 0.258756
stepper_z2 = -0.006820
stepper_z3 = 0.038193
18:03:03 - Average: 8.409687
18:03:03 - Actuator Positions:
z: 8.699815 z1: 8.150931 z2: 8.416507 z3: 8.371494
18:03:03 - Gantry-relative probe points:
0: 8.611127 1: 8.288298 2: 8.382844 3: 8.423270
18:02:57 - Retries: 0/5 Probed points range: 2.489502 tolerance: 1.000000
18:02:57 - Making the following Z adjustments:
stepper_z = -0.688046
stepper_z1 = 2.569135
stepper_z2 = -0.672008
stepper_z3 = -1.209080
18:02:57 - Average: 8.334078
18:02:57 - Actuator Positions: 
z: 9.022124 z1: 5.764943 z2: 9.006086 z3: 9.543158
18:02:57 - Gantry-relative probe points: 
0: 8.906120 1: 6.891709 2: 8.660230 3: 9.381210
18:02:26 - Bed heating...
18:02:26 - The bed target temperature was not reached!
18:02:26 - Check Heating!
18:02:26 - Leveling gantry...
18:02:26 - Waiting for bed...
18:02:26 - Result is z=1.950289
18:02:13 - Homing Z
18:01:36 - Cleaning nozzle...
18:01:36 - Result is z=1.986093
18:01:10 - Homing Z
18:01:10 - Homing X
18:01:10 - Homing Y
18:01:10 - Heating bed...
18:01:09 - File selected
18:01:09 - File opened:EM_Test_v2_PLA-T_0.2_40m40s.gcode Size:1600814
15:40:02 - Done printing file
15:39:59 - Finish Print!
```
The toolhead was in a different position before the home as cancelling the print moves the toolhead to the corner.

vvuk, 2025-03-10:
Ok, so:

- cancel print
- printer goes idle/unhomed (I think cancel just automatically turns off the steppers too)
- home
- nozzle clean
- QGL
- tap?

and that one fails? Is it happening at the "qgl end" point, one of the corners?

Coare1, 2025-03-10:
Almost, just finishing the print, not cancelling.

I move the nozzle to the center point (117.5, 117.5, 5) and then tap. This is the same location homing puts me at at the end of the homing process. I think this could be an issue with not using the Eddy for a while (would be weird as it uses it during homing z, but I don't know), so I'll try doing something like a PEPS before the tap.

vvuk, 2025-03-10:
One thing that's different about homing vs tap is that homing is more lenient about some errors; it ignores one type of error because that error also happens when the sensor is very far from the build plate. tap aborts at any error though for safety, though you can set `max_errors: ` to some value in the config to allow it to ignore that many errors in a row. But it's still strange that it just doesn't work at all the first tap, then works fine.

Coare1, 2025-03-10:
Yeah I set the max errors to 10 just to see if it would help but nothing worked.

thhdragon, 2025-03-16:
might this be related to the accelerometer on the toolhead board giving an error the first read after bootup? happens on my sht36 v3 and at one point kusba had an error first read too. like its unloaded until it calls it

JHFexafuse, 2025-09-25:
I have a kind of similar issue. I am running 20 eddys, out of wich they all work fine after the initial setup, but they fail to tap after some use. The its a matter of clearing all config data and doing the setup again, for some that works, but others need wma with a threshold as low as 300. I feel like im missing something, because theres no logic to follow with the issue.

danielfse2002-beep, 2025-10-05:
Anyone solve this issue yet I've tried changing the target_z like klipper says but can't seem to find where, even wrote a macro to include the variable_ target_z but still to no avail, 
Strangely enough when I make to command to tap it works fine but when I try to start a print it fails

vvuk, 2026-11-01:
`develop` has a fix for stale sensor data being handed to the next operation. That would match "the first tap fails, then everything works", especially since running `PEPS` first hides it. If you can, try `develop` and let me know.