vvuk/eddy-ng #32: [Suggestion] Auto fallback for drive current on heat-warped beds (e.g. Sovol SV08)

benjhess, 2025-04-01:
I recently switched to eddy-ng on my SV08, and it took quite a bit of time to get everything running reliably. I kept running into issues with failed TAPs and inconsistent probing results, mostly caused by incorrect drive current settings and the SV08's notoriously warped bed. The Sovol SV08 has a noticeable taco-shaped warp, which gets worse the longer and hotter it runs. As a result, the Eddy sensor’s effective probe range tends to shift during operation.

If the drive current doesn’t match the current bed condition, probing (using PROBE_EDDY_NG_PROBE_STATIC) often fails with an error like: `No samples received for time 89.968 (waited for 0.348)`

Of course, it’s possible to calibrate for different drive currents and then switch drive current manually, which is what i did so far:

```
# calibration for bed temp 60c and heat soak for at least 30 mins (pla)
PROBE_EDDY_NG_CALIBRATE DRIVE_CURRENT=16

# calibration for bed temp 100c and heat soak for at least 40 mins (abs, asa, etc.)
PROBE_EDDY_NG_CALIBRATE DRIVE_CURRENT=17
```

Macros to switch to the one, that currently works:

```
[gcode_macro SET_EDDY_DRIVE_CURRENT_TO_LOW]
gcode:
  LDC_NG_SET_DRIVE_CURRENT CHIP=btt_eddy VAL=16

[gcode_macro SET_EDDY_DRIVE_CURRENT_TO_HIGH]
gcode:
  LDC_NG_SET_DRIVE_CURRENT CHIP=btt_eddy VAL=17
```

But i think, this is not really an ideal solution^^

So instead of adjusting the drive current manually every time probing fails, I built a small Python module that automatically cycles through different current values until it finds one that works.

adaptive_drive_current.py:

```
import logging

class AdaptiveDriveCurrent:
    def __init__(self, config):
        self.printer = config.get_printer()
        self.gcode = self.printer.lookup_object('gcode')
        self.logger = logging.getLogger('adaptive_drive_current')
        
        # Get configuration options
        self.default_currents_str = config.get('default_currents', '17, 16')
        
        # Parse default currents
        try:
            self.default_currents = [int(c.strip()) for c in self.default_currents_str.split(',')]
            if not self.default_currents:
                raise ValueError("Empty current list")
        except Exception as e:
            self.logger.error("Error parsing default_currents: %s", str(e))
            self.default_currents = [17, 16]  # Fallback to hardcoded defaults
        
        self.logger.info("Default currents: %s", self.default_currents)
        
        # Get chip name configuration
        self.chip_name = config.get('chip_name', 'btt_eddy')
        self.logger.info("Using chip name: %s", self.chip_name)
        
        # Register our main command and helper commands
        self.gcode.register_command(
            'ADAPTIVE_DRIVE_CURRENT',
            self.cmd_ADAPTIVE_DRIVE_CURRENT,
            desc="Find and set working drive current for probing"
        )
        
        self.gcode.register_command(
            '_TRY_WITH_NEXT_CURRENT',
            self.cmd_try_with_next_current,
            desc="Try probing with next current value"
        )
        
        # Current state
        self.currents = []
        self.current_index = 0
        
        self.logger.info("Adaptive drive current module initialized")
    
    def cmd_try_with_next_current(self, gcmd):
        """Try probing with the next current value"""
        self.current_index += 1
        
        if self.current_index >= len(self.currents):
            # We've tried all current values
            self.gcode.respond_info("All current values failed")
            return
        
        current = self.currents[self.current_index]
        self.logger.info("Trying with current %d", current)
        self.gcode.respond_info(f"Previous current failed, trying with current: {current}")
        
        try:
            # Set the current
            self.gcode.run_script_from_command(f"LDC_NG_SET_DRIVE_CURRENT CHIP={self.chip_name} VAL={current}")
            
            # Try probing
            self.gcode.run_script_from_command("PROBE_EDDY_NG_PROBE_STATIC HOME_Z=1")
            
            # If we get here, it worked
            self.gcode.respond_info(f"Probing successful with current {current}")
        except Exception as e:
            # This current also failed
            error_msg = str(e)
            self.gcode.respond_info(f"Probing failed with current {current}: {error_msg}")
            
            # Try the next current value
            self.cmd_try_with_next_current(gcmd)
    
    def cmd_ADAPTIVE_DRIVE_CURRENT(self, gcmd):
        """Find and set working drive current for probing"""
        # Get currents from parameter (comma-separated list)
        currents_param = gcmd.get("CURRENTS", None)
        chip_param = gcmd.get("CHIP", None)
        
        # Override chip name if provided
        if chip_param:
            self.chip_name = chip_param
        
        if currents_param:
            # Parse the comma-separated currents
            try:
                self.currents = [int(c.strip()) for c in currents_param.split(",")]
                self.logger.info("Using specified currents: %s", self.currents)
            except Exception as e:
                # Invalid format, use defaults
                self.logger.error("Invalid currents format: %s", str(e))
                self.currents = self.default_currents.copy()
                self.logger.info("Using default currents: %s", self.currents)
        else:
            # Use default currents
            self.currents = self.default_currents.copy()
            self.logger.info("Using default currents: %s", self.currents)
        
        # Reset the index
        self.current_index = 0
        
        if not self.currents:
            self.gcode.respond_info("No current values specified")
            return
        
        # Start the adaptive process
        self.logger.info("Starting adaptive drive current adjustment")
        current = self.currents[0]
        self.gcode.respond_info(f"Trying with drive current: {current}")
        
        try:
            # Set the first current value
            self.gcode.run_script_from_command(f"LDC_NG_SET_DRIVE_CURRENT CHIP={self.chip_name} VAL={current}")
            
            # Try probing
            self.gcode.run_script_from_command("PROBE_EDDY_NG_PROBE_STATIC HOME_Z=1")
            
            # If we get here, the first current worked
            self.gcode.respond_info(f"Probing successful with current {current}")
        except Exception as e:
            # First current failed
            error_msg = str(e)
            self.logger.error("Probing failed with current %d: %s", current, error_msg)
            self.gcode.respond_info(f"Probing failed with current {current}: {error_msg}")
            
            # Try with next current values
            self.cmd_try_with_next_current(gcmd)

def load_config(config):
    return AdaptiveDriveCurrent(config)
```

Copy python script to ~/klipper/klippy/extras/adaptive_drive_current.py
Add somewhere to config (eddy.cfg, printer.cfg, whatever):

```
[adaptive_drive_current]
# currents to try. note: high to low is better here. 
# probe static might work but tap will eventually still fail!
default_currents: 17, 16 
chip_name: btt_eddy
```

Now for homing override, instead of `PROBE_EDDY_NG_PROBE_STATIC HOME_Z=1`, you can use this, which automatically cycles through different current values until it finds one that works:
`ADAPTIVE_DRIVE_CURRENT`

You can also override defaults when running manually:
`ADAPTIVE_DRIVE_CURRENT CURRENTS=18,17,16 CHIP=my_eddy`

This might be a good addition to the docs. Especially for users whose printer beds tend to deform under heat, causing inconsistent probe behavior :)

Btw: Thanks a lot for your amazing work on eddy-ng. It’s an awesome upgrade and is so much more accurate than temp compensation :)

samschultzponsys, 2025-04-14:
This is fantastic, do you find this to work well enough for a user to pick it up or should we wait for it to be either integrated into eddyng or not?

benjhess, 2025-04-14:
@samschultzponsys 
Thanks :) Most of the time this solution works really well. However, I’ve since realized that the problem isn’t just the warped bed on the SV08. The Eddy probe itself appears to be quite temperature-sensitive. The hotter the probe’s MCU gets, the more the required drive current drifts upwards.

For example, I recently tried to print ABS with a long heat soak (around 30 minutes), and the toolhead was positioned quite close to the bed. The Eddy probe’s MCU reached around 70 °C, and at that point I couldn’t get a single TAP to work, not even by calibrating a new drive current. I had to move the toolhead far away from the bed, let the MCU cool down to ~45 °C, and then TAP started working again.

To test this independently of the bed, I used a heat gun to warm up the probe’s MCU, and sure enough... the higher the temperature, the higher the required drive current. At 70 °C and above, probing completely fails for me, regardless of current settings.

So I’ve updated my heat soak macro to ensure the toolhead stays at least +50 mm above the bed during the soak phase. That way, the probe’s MCU doesn’t heat up too much, and I haven’t seen the issue since.

All in all, it seems that temperature drifting is definitely still an issue with eddy-ng, just like it was with the original setup. That said, as long as a TAP is possible, eddy ng delivers significantly more precise results compared to the original implementation.

For me it currently works well enough, but it would definitely benefit from more users testing it. Feel free to give it a try and let me know how it works out for you.

vvuk, 2025-04-14:
Hey, thanks for doing this -- I'll take a look at how to incorporate something like this directly, though it might take me a week or two (in the middle of a house move :).  Note that the MCU temp shouldn't be relevant, however the temperature of the coil itself is. The MCU temp can be a good proxy for that though. Also this is largely a problem with the BTT Eddy more so than the others, due to the coil design.

I print ABS/ASA often with a chamber of around 40C+ and with long 100C bed heat soaks. I usually park my toolhead in the back left and (I think) 30-40mm off the bed while that's happening and things generally work. This worked with the BTT Eddy, though I've moved on to using Mellow's probe that's directly attached to their toolboard.

> All in all, it seems that temperature drifting is definitely still an issue with eddy-ng, just like it was with the original setup. That said, as long as a TAP is possible, eddy ng delivers significantly more precise results compared to the original implementation.

It is, that's just a physical fact, but there are two separate (but related) impacts of temperature:

1. "temperature drift" where the computed height values don't match reality based on calibration due to changing physical parameters, which is what "temperature correction" tries to fix for
2. the coil parameters drifting out of spec due to temps, where the probe can't get any frequency readings at all (even ignoring the height conversion)

The second is always an issue; it's just less of an impact on the BTT Eddy with the "stock" setup, because the probe is always much higher off the plate (2.5mm in the default setup), but then you have the first issue to deal with in order to get accurate height readings.

For eddy-ng, the first issue is negated entirely by tap (because calibration values are irrelevant with tap, all that matters is that the sensor can provide raw data), but the second becomes something that needs to be balanced.

In typing this out I also just realized something... there's actually no need to require calibration for the "tap" drive currents, as those converted height readings aren't used for anything. I'll think about removing the need for that as well.

samschultzponsys, 2025-04-15:
@vvuk @benjhess Sounds like you two have a good plan here, and if I can get my eddy working at all I would be very interested to help test this setup. All we need now is a 4x4mm noctua fan and we should be all set haha

benjhess, 2025-04-15:
@vvuk 
Thanks for the clarification :) That makes a lot of sense. You're right, I should have said coil temperature instead of MCU temp. I just used the MCU temp as a rough proxy during my tests, since it’s the only temperature I could easily observe. The strong dependency on temperature seems to be particularly noticeable with the BTT Eddy, as you said.

I agree with your breakdown of the two temperature effects. In my case, it was the second one where the coil gets too hot to return valid frequency data at all. That’s been the remaining issue after integrating adaptive drive current. Good to know that this is mostly due to the BTT Eddy’s design.

Also really interesting insight about possibly removing the need to calibrate drive currents for TAP, since the absolute height isn’t needed. That would definitely simplify things.

Thanks again for all your work on this. Really appreciate it!