vvuk/eddy-ng #34: QGL crashes firmware with current develop version

thatguyfromms, 2025-04-07:
[klippy.log](https://github.com/user-attachments/files/19636750/klippy.log)

Hi vlad,
first of all: tremendous work so far, this version is so much better than mainline klipper  one, hope you will somehow convince Kevin to integrate it.

With the current develop-version (commit 2c8a1f8) homing works fine, but after issuing a 
QUAD_GANTRY_LEVEL_BASE horizontal_move_z=5 retry_tolerance=1
it scans the four corners and then immediately crashes.
Mainsail: Internal error on command:"QUAD_GANTRY_LEVEL_BASE"
klippy.log: serialhdl.error: Unable to obtain 'sensor_bulk_status' response
klippy.log attached.

Reverting to beedce8 fixes the issue.

If you need more info, let me know.

vvuk, 2025-04-07:
Thank you! I just landed a bunch of code on develop so this is very timely. Note the "Unable to obtain sensor_bulk_status" error is always a red herring, klipper almost always bails with that if a bulk sensor operation is in progress.  The actual error is earlier in the log:

```
Internal error on command:"QUAD_GANTRY_LEVEL_BASE"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/biqu/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/biqu/klipper/klippy/gcode.py", line 230, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/biqu/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/quad_gantry_level.py", line 53, in cmd_QUAD_GANTRY_LEVEL
    self.probe_helper.start_probe(gcmd)
  File "/home/biqu/klipper/klippy/extras/probe.py", line 459, in start_probe
    results = probe_session.pull_probed_results()
  File "/home/biqu/klipper/klippy/extras/probe_eddy_ng.py", line 2254, in pull_probed_results
    height = self._sampler.find_height_at_time(start_time, end_time)
  File "/home/biqu/klipper/klippy/extras/probe_eddy_ng.py", line 2774, in find_height_at_time
    f"find_height_at_time: {len(self.times)} samples, time range {self.times[0][0]:.3f} to {self.times[-1][0]:.3f}"
TypeError: 'float' object is not subscriptable
```

I'll fix shortly!

thatguyfromms, 2025-04-07:
Home, QGL, TAP, Bed_Mesh: Working. 👍
And you were not kidding with "speeding up calculations". A 30x30 mesh is now done in what feels less than a second (SV08 mainboard). Very impressive.
Thank you!

commit da038044 (2025-04-07): Fix bad print in find_height_at_time (fixes #34)
closed 2025-04-07 by vvuk