vvuk/eddy-ng #35: TAP does not apply z-offset in startup g-code

Erop1928, 2025-04-12:
I updated the clipper on 10.04.25, then updated eddy-ng and flashed all the mcu. However, now at the beginning of printing PROBE_EDDY_NG_TAP does not apply z-offset.

Here is my starting g-code
```
M140 S[first_layer_bed_temperature]
M104 S180
M106 S0

MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! FILAMENT_NAMES=!filament_names! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK
MMU_START_LOAD_INITIAL_TOOL

M190 S[first_layer_bed_temperature]
M109 S150
G28
PROBE_EDDY_NG_TAP ; THRESHOLD=600
M104 S[first_layer_temperature] 
BED_MESH_CALIBRATE METHOD=rapid_scan 

G90
G1 X5 Y5 Z0.5 F10000.0 ; go intro start pos
G1 Y200.0 E50.0 F600.0 ; intro line to prime extruder
G1 X5.5 Z0.8 E0.2 F800.0 ; intro line
G1 Y5 E80 F800.0 ; intro line
G92 E0.0 ; reset extruder distance position
```

[image]

vvuk, 2025-04-12:
Hmm, what do you mean by "doesn't apply z offset"? If you're upgrading from an older version of eddy-ng the behaviour changed to home the Z axis on tap, meaning you'll see a z offset of 0 right after a tap. Stepping Z after that and saving will correctly save an updated tap_adjust_z value.

Erop1928, 2025-04-12:
After using PROBE_EDDY_NG_TAP, when starting printing, the z-offset value does not change in the block. And the printer continues printing without offset. If you then apply a manual offset via the web interface to the value calculated as a result of TAP, then you get a great first layer.

vvuk, 2025-04-12:
Right -- with the latest code, the Z offset displayed after tap will always be 0.000.  The Z axis will be homed so that Z=0 is at the tap value, which is the detected tap height + the `tap_adjust_z` configuration value.  You need to tweak tap_adjust_z, but you can do that by manually stepping after tap like you did, and then pressing the save button to save it. That will update `tap_adjust_z`, and the next time you do a tap, it will have that additional amount already accounted for.

haski369, 2025-04-16:
Do I need to reflash the eddy probe with Eddy-ng each time I update Klipper??? Is that the reason I get an error Fluidd saying Unknown pin chip name 'probe'??

vvuk, 2025-04-20:
> Do I need to reflash the eddy probe with Eddy-ng each time I update Klipper??? 

This is more of a klipper question -- you may need to, you may not. It's no different than any other MCUs. Usually safest to flash everything after every update.

> Is that the reason I get an error Fluidd saying Unknown pin chip name 'probe'??

No, this is something in your config. I don't have any ideas what though, sorry!

closed 2025-04-20 by vvuk

haski369, 2025-04-20:
`No, this is something in your config. I don't have any ideas what though, sorry!`
It was Eddy probe that I had to reflash after the update. After reflashing the problem went away! :)