vvuk/eddy-ng #38: Unable to complete calibration

samschultzponsys, 2025-04-15:
My printer was restart during this process:

Move the toolhead to approximately 10mm off the bed. (Note: if you are unable to home Z at all, do a manual Z home first.)
Run LDC_NG_CALIBRATE_DRIVE_CURRENT CHIP=[your chip name] (your chip name is the ... part of [probe_eddy_ng ...] in your config)
Take note of the value it shows.
Run PROBE_EDDY_NG_TEST_DRIVE_CURRENT DRIVE_CURRENT=[that value]. Take note of the height range. If it includes the 1mm..4mm range, this is a good drive current for homing. Set reg_drive_current: ... to this value in your config. If the height goes all the way down to 0.0xx, this should be a good drive current for tap operations as well. In that case, stop here. If it doesn't go down to 0.0xx, continue.
If the range that's returned is higher than the range needed for home and/or tap, run again with DRIVE_CURRENT= 1 higher than previously. If the range returned is lower, run with 1 lower.
Try to find an appropriate drive current for homing and tapping. When you find a drive current for tap that's different than the homing one, set tap_drive_current: ... in the config.
If you're unable, consider moving the sensor higher or lower.

I was able to get an intial reading of 24, and the range was within the recommended tolerance on my Sovol SV08. After the printer restart I wanted to continue calibration but no calibration commands are working, all the commands want me to home x and y but I cannot home x and y, I just get error: 
$ G28 X
20:03:08 
!! Drive current 15 not calibrated
20:03:08 
!! Drive current 15 not calibrated

I would like to just restart the calibration process completely but I am unsure how to do so. I have actually completely wiped my printers EMMC and started with fresh mainline klipper 13 and I am still getting stuck on the calibration values. Any assistance is greatly appreciated, I was also unable to find any troubleshooting section on the git outside of the FAQ

marcams, 2025-04-17:
I am encountering the same issue after upgrading Klipper to version 13. The printer was functioning properly with the BTT Eddy and Eddy_NG software. I followed the upgrade procedure outlined in the wiki, but it is now displaying the error "Drive current 15 not calibrated."

samschultzponsys, 2025-04-22:
I will update mine shortly, I just found out someone scammed amazon and I ended up getting a returned eddy not eddy duo in a box. I have the new one and will be attempting this now.

thhdragon, 2025-04-23:
i did this 
cd ~/eddy-ng
git pull
./install.sh # only if Klipper
and then redid setup and it started working fine on 13

mghenderson, 2025-05-02:
@samschultzponsys Did you ever find a fix for this?

samschultzponsys, 2025-05-02:
Yes so sorry! It turned out I just need to actually have the correct hardware, I have noticed any time I change nozzles or hotend parts I have to delete everything at the bottom of my printer cfg section, re-run the probe setup and then go from there.

closed 2025-05-02 by samschultzponsys

vvuk, 2025-05-02:
At some point I'm going to implement "profiles" so that you can save different nozzles/build plates/etc which should make that simpler!

samschultzponsys, 2025-05-02:
@vvuk that is fantastic, in my case I literally have been replacing broken parts, so that is where that comes in. At least the steps to get it working are solid. Your tool works like a champ and I am on Klipper .13