vvuk/eddy-ng #39: Lost communication with MCU 'eddy'

zimmra, 2025-04-15:
Recently updated eddy-ng, up until yesterday I had been running 3cebe44 (and previous versions) with essentially no issues. 

Updated to latest, then reflashed my main MCU, toolhead MCU, and Eddy MCU afterwards

It didn't like my previously saved setup/calibration, had to re-run `PROBE_EDDY_NG_SETUP` to recalibrate

The first two prints in a row after performing these steps I am getting an error mid-way through the print that I have never got before. 

```
Lost communication with MCU 'eddy'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

[image]

I verified my USB cable is fully seated on both the eddy end and the mainboard side.

I'm still leaning towards it having something to do with the USB cable, but the timing of it working for over a month and then suddenly failing right after being updated is strange so I am wanting to check in here 

Here are the last lines prior to the crash from my klippy logs, nothing seems 'wrong' with them, the eddy timeout seems sudden

```
Stats 6151.5: gcodein=0  mcu: mcu_awake=0.132 mcu_task_avg=0.000044 mcu_task_stddev=0.000089 bytes_write=11434627 bytes_read=3174376 bytes_retransmit=9 bytes_invalid=0 send_seq=242913 receive_seq=242913 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=48 upcoming_bytes=5086 freq=72004151 eddy: mcu_awake=0.001 mcu_task_avg=0.000003 mcu_task_stddev=0.000002 bytes_write=61528 bytes_read=849522 bytes_retransmit=9 bytes_invalid=0 send_seq=8578 receive_seq=8578 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=4 freq=12000524 adj=11999855 extra_mcu: mcu_awake=0.014 mcu_task_avg=0.000035 mcu_task_stddev=0.000034 bytes_write=1878872 bytes_read=1231411 bytes_retransmit=9 bytes_invalid=0 send_seq=47425 receive_seq=47425 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=72004482 adj=72000371 sd_pos=2018505 Eddy_MCU_temp: temp=86.4 Eddy_temp: temp=84.7 heater_bed: target=95 temp=95.0 pwm=0.240 main_MCU_temp: temp=47.3 CPU_temp: temp=63.2 toolhead_MCU_Temp: temp=84.4 sysload=0.59 cputime=678.680 memavail=601780 print_time=6162.847 buffer_time=2.103 print_stall=0 extruder: target=260 temp=260.7 pwm=0.358
Timeout with MCU 'eddy' (eventtime=6152.552377)
Transition to shutdown state: Lost communication with MCU 'eddy'
```

Running klipper v0.12.0-486, moonraker v0.9.3-72

vvuk, 2025-04-15:
can you upload your full klippy log? The useful info will be much earlier unfortunately

zimmra, 2025-04-16:
Sure, here is the log trimmed up to just one of the failed prints

[failed-print-klippy.2025-04-15.log](https://github.com/user-attachments/files/19785825/failed-print-klippy.2025-04-15.log)

FWIW, I did have a successful print after those first 2 failed (I did another shutdown and this time completely reseated the eddy's cable instead of verifying it was pushed all the way in, and did a much shorter print than the failures). I have another print that's about 4 hours in that's going fine right now too. So perhaps the failures were a fluke or perhaps an intermittent/inconsistent issue.

vvuk, 2025-04-20:
Yeah, this looks like a cable issue or other physical issue; there's nothing in the logs indicating a crash of any kind. One thing I did notice -- you should consider changing your bed mesh config to use `horizontal_move_z: 2.0`. It looks like it's set to 3.0 right now; 2.0 will be more accurate.

closed 2025-04-20 by vvuk