vvuk/eddy-ng #40: EDDYng: Tap failed: 15 errors, last: Error during homing probe: Sensor error (Over-range Error Watchdog Error) at toolhead z=-0.355

wysockiw, 2025-04-18:
Everything was fine before update to latest version ... 

Tap is working just fine when doing intial TAP - simply running G28 XYZ and PROBE_EDDY_NG_TAP ... but when trying to print anything then it is throwing error 

EDDYng: Tap failed: 15 errors, last: Error during homing probe: Sensor error (Over-range Error Amplitude Error) at toolhead z=-0.225

here is my START_PRINT macro 

    SET_FAN_SPEED FAN=hotend_fan SPEED=0.6                                    # Set hotEnd fan to min 60% 
    SET_DISPLAY_TEXT MSG="Set Bed temp to {BED_TEMP}c and continue ... "      # Displays info
    RESPOND TYPE=command MSG='Set Bed temp to {BED_TEMP}c and continue ... ' 
    M140 S{BED_TEMP}                                                          #Set the Bed tem but continue don't wait  
    
    QUAD_GANTRY_LEVEL
    
    # Move to the dirt bin and wait for temp 
    G0 Z10 X20 Y354 F10000    

    # Nozzle tem initial temp and wait 
    SET_DISPLAY_TEXT MSG="Set nozzle temp to 160c and wait ... "              # Displays info
    RESPOND TYPE=command MSG='Set nozzle temp to 160c and wait ... ' 
    M109 S160                                                                 # Set extruder temp for the clean up and wait 

    # Clean Nozzle 
    RESPOND TYPE=command MSG='Going to Clean nozzle  ... ' 
    CLEAN_NOZZLE
    SET_GCODE_VARIABLE MACRO=PRINT_START VARIABLE=nozzle_clean VALUE=1

    # Got to center and do TAP 
    G0 Z10 X175 Y185
    PROBE_EDDY_NG_TAP

wdtamagi, 2025-05-13:
Did you find a way to fix it? I'm having the same problem.

lanpartyceo, 2025-06-23:
I found a way for my self that has been working so far.

You can manually set the tap and the reg drive current in the config file. tap for tap, reg for homing and calibration.

so set both to 15, calibrate, save. set both to 16, calibrate, save. now i have 15 and 16 calibrated and saved. then i set 15 for the reg and 16 for the tap, and so far.... is moving better.

It seems what current setting is good for one action is bad for the other and vice versa. I have been able to successfully start multiple print jobs (tests and otherwise) without the error re-occurring after doing this

[image]

vvuk, 2025-06-23:
> so set both to 15, do manual calibration, save. set both to 16, do manual calibration, save. now i have 15 and 16 calibrated and saved. then i set 15 for the reg and 16 for the tap, and so far.... is moving better.

`PROBE_EDDY_NG_CALIBRATE DRIVE_CURRENT=15` or =16 or whatever will do a calibration at whatever the drive current you specify, no need to change the value in the config in order to calibrate at a particular value.