vvuk/eddy-ng #43: Doc: Homing Override for eddy-ng

SmileyMan, 2025-04-23:
Extends the Z macro in Wiki.

```
[homing_override]
axes: xyz
gcode: 
    M117 Homing Override
    # Used for completion message
    {% set homing_success = 1 %}   
    
    # Looks for XY in params and homes Y first - Comment out to disable
    {% if not rawparams or ('Y' in rawparams and 'X' in rawparams) %}
      M117 Homing Y
      G28 Y F1000
    {% endif %}
    
    # X in params so home X
    {% if not rawparams or 'X' in rawparams %}
      M117 Homing X
      G28 X
    {% endif %}

    # Y in params so home Y
    {% if 'Y' in rawparams and 'X' not in rawparams %}
      M117 Homing Y
      G28 Y F1000
    {% endif %}
        
    # Check status of axes XY in printer variables or exists in params or params empty before homing Z
    {% if not rawparams or 'Z' in rawparams %}
      {% if
        not rawparams or 
        (
          rawparams and 
          (
            ("x" in printer.toolhead.homed_axes or 'X' in rawparams) and 
            ("y" in printer.toolhead.homed_axes or 'Y' in rawparams)
          )
        ) 
      %}
        # Set printer probe adjustments for centering probe on bed
        {% set probe_offset_x = 0 %}
        {% set probe_offset_y = 20 %}
 
        # Get printer xy limits
        {% set printer_min_x = printer.toolhead.axis_minimum.x | int %}   
        {% set printer_min_y = printer.toolhead.axis_minimum.y | int %}   
        {% set printer_max_x = printer.toolhead.axis_maximum.x | int %}   
        {% set printer_max_y = printer.toolhead.axis_maximum.y | int %}

        M117 Centering on bed
        # Using max and min to account for deltas with negitive axis
        G1 X{((printer_max_x / 2) + (printer_min_x / 2)) - probe_offset_x} Y{((printer_max_y / 2) + (printer_min_y / 2)) - probe_offset_y} F4800
        G4 S1                                            ; chill for a sec
        M400                                             ; wait for move to finish
        M117 Homing Z
        G28 Z                                            ; do the coarse home
        G90                                              ; set absolute positioning
        G0 Z2 F1000                                      ; to 2mm -- home height (or whatever you'd like), for maximum sensor accuracy
        G4 S1                                            ; chill for a sec
        M400                                             ; wait for move to finish
        M117 Reading height
        PROBE_EDDY_NG_PROBE_STATIC HOME_Z=1              ; read the current exact height from sensor and home Z to it
        G0 Z20 F1000                                     ; to 20mm as home
        G4 S1                                            ; chill for a sec
        M400                                             ; wait for move to finish
        # Center head on bed
        G1 X{(printer_max_x / 2) + (printer_min_x / 2)} Y{(printer_max_y / 2) + (printer_min_y / 2)} F4800
      {% else %}
        {% set homing_success = 0 %}
        M117 X and Y axis must be already homed
      {% endif %}
    {% endif %}
    
    {% if homing_success == 1 %}
      M117 Homing completed
    {% else %}
      M117 Homing failed
    {% endif %}
```

maj0r1337, 2025-05-26:
Hi @SmileyMan ,
ive tried this override but it fails

15:57:57
Must home axis first: 295.426 294.539 0.016 [0.000]
15:57:57
Must home axis first: 295.426 294.539 0.016 [0.000]
15:57:51
G28

this happens when i press the home all button in mainsail

vvuk, 2025-05-27:
Hm can you describe what the purpose of the changes is? It looks like it just moves to the center of the bed? Also, at a quick glance, the probe offsets should be read directly from the 'probe' object vs. specified inline here, but there isn't really any reason to do this; it's technically more precise, but the precise Z offset should be set by tap later anyway.

closed 2025-07-16 by vvuk