vvuk/eddy-ng #44: Issue: PROBE_EDDY_NG_SET_TAP_ADJUST command not recognized.

greatguitartist, 2025-04-24:
When I run the PROBE_EDDY_NG_SET_TAP_ADJUST VALUE=0.08 (for my printer), I get "Unknown Command PROBE_EDDY_NG_SET_TAP_ADJUST". I have verified everything in the installation per the wiki guide, so I'm not sure what's missing.

Additionally, I have plotly installed, but I don't see a tap.xml file in the /tmp folder.

greatguitartist, 2025-04-24:
Looked at the probe_eddy_ng.py file and saw that the command to set the tap offset is in fact PROBE_EDDY_NG_SET_TAP_ADJUST_Z and not PROBE_EDDY_NG_SET_TAP_ADJUST. Documentation needs to be updated.

Still having issue with no tap.xml file in the /tmp folder, though

vvuk, 2025-04-24:
> Looked at the probe_eddy_ng.py file and saw that the command to set the tap offset is in fact PROBE_EDDY_NG_SET_TAP_ADJUST_Z and not PROBE_EDDY_NG_SET_TAP_ADJUST. Documentation needs to be updated.

Whoops! Sorry about that. I'll fix.

In /tmp, there will be `tap-samples.csv`, and `tap.html` if you have `write_tap_plot: true` (and optionally `write_every_tap_plot: true`) in the config. I'm behind on doc updates, I changed the default to "false" because in some cases there's a bit of a delay on slower hosts for the first tap, as python loads up all the plotting libs.

You can also pass `PLOT=1` (just one) or `PLOT=2` (every sample) as arguments to TAP to do it on-demand.

vvuk, 2025-04-24:
> Looked at the probe_eddy_ng.py file and saw that the command to set the tap offset is in fact PROBE_EDDY_NG_SET_TAP_ADJUST_Z and not PROBE_EDDY_NG_SET_TAP_ADJUST

Wait, where do you see `PROBE_EDDY_NG_SET_TAP_ADJUST`? I don't see any references to that in the readme or wiki.

greatguitartist, 2025-04-24:
[image]

greatguitartist, 2025-04-24:
No worries. Thanks for the quick response. I'll set the config to true and try. Thanks again!

closed 2025-04-24 by greatguitartist

vvuk, 2025-04-24:
> [image]

🤦‍♂ my fingers searched for `TAP_ADJUST_Z` :) thanks!