vvuk/eddy-ng #49: Option 'mesh_min' in section 'bed_mesh' must be specified

chartjes, 2025-05-11:
When following the instructions to install eddy-ng with a BTT USB Duo, I get the following error after adding in the suggested items to my `printer.cfg`

`Option 'mesh_min' in section 'bed_mesh' must be specified`

N3rdyBlacksmith, 2025-05-22:
I just started tinkering with Eddy-NG and I've got it mostly working.  I had to add this under the [bed mesh] heading.  It's working, but I need to define it a little more accurately so that it gets more of the bed.  The #'s are what the original values were.

mesh_min: 20,21.5 #10,10
mesh_max: 330,331.5 #333,340

vvuk, 2025-06-08:
This looks like a misconfigured `bed_mesh` -- eddy-ng doesn't require bed_mesh so this is part of your other config. Possibly bed_mesh was deleted without deleting bed_mesh entries from the saved data section at the end of printer.cfg?

closed 2025-06-08 by vvuk

N3rdyBlacksmith, 2025-06-09:
Yep.  That was the case.  I can't believe I missed that.  Thanks again for your help,

-Steve