vvuk/eddy-ng #50: Use eddy-NG on multitool printer

den-kozlov, 2025-05-18:
Hi! Thank for the project!

The tapping works great for every tool of my multitool printer. But this limits me on using just one tool during the print. Of course I can measure relative z-offsets for all my tools (relative to the first one) and apply corresponding offset on every tool change. But I would like to measure offset individually for every tool on print start. I think that could give more precise z-offset since extruder temperatures could be different during multi-material print.

Hence the question - How can I get calculated z-offset after tap to save it and restore on tool change?

vvuk, 2025-06-08:
To verify: you have a sensor installed somewhere where you can get per-tool Z data? You can just do a PROBE manually, or use TAP with `HOME_Z=0` in which case it will set the gcode offset and won't re-home the Z axis, and then you can do something with the gcode offset. Is that what you're looking for?

vvuk, 2025-07-16:
So ultimately I think what you want is something like a `PROBE METHOD=tap` where it doesn't change any homing or do anything else, just saves the tap-Z in the last probed status field?

vvuk, 2026-11-01:
On `develop`, probe commands accept `METHOD=tap`, which taps at the current position without re-homing Z or changing the G-code offset. The result is in `printer.probe.last_probe_position.z`. So per tool you can `PROBE METHOD=tap`, store that value in a variable, and apply it with `SET_GCODE_OFFSET` on tool change. See "Probing with tap" in `docs/tap.md`.

closed 2026-11-01 by vvuk