vvuk/eddy-ng #55: Add mainsail update script

brandonb1987, 2025-05-27:
Add in the following to the instructions so others can simply add it to their moonraker update tracker.

Add the following to moonraker.conf:

`[update_manager eddy-ng]
type: git_repo
path: ~/eddy-ng
origin: https://github.com/vvuk/eddy-ng.git
managed_services: klipper
primary_branch: main
install_script: install.sh'

brandonb1987, 2025-05-27:
I see the issue now. Every time you push for updates it will prompt as dirty and be overwritten

closed 2025-05-27 by brandonb1987

vvuk, 2025-05-27:
Every time klipper gets updated you mean? Yeah, I wish the update manager wasn't as aggressive about resetting state, or if there was a way to tell it "preserve changes to these files". I actually have an idea about how to do things without requiring changes to bed_mesh, which is the main thing that needs to stay; the other changes (to the Makefile) are OK to get blown away after the firmware install is done (they have to come back if there's a firmware update, but aren't needed during normal running)