vvuk/eddy-ng #57: QGL not working

NickDK1, 2025-06-01:
the scan type qgl dodsnt work for me at all it wants to make  super big adjudsments.

and gives errors like .

Aborting quad_gantry_level required adjustment 878.316638 is greater than max_adjust 30.000000
23.36
Average: -127.352693
23.36
Actuator Positions:
z: -1005.669331 z1: 265.986675 z2: -73.479202 z3: 303.751086
23.36
Gantry-relative probe points:
0: -664.100383 1: 4.244752 2: 4.183075 3: 7.793780
23.35
Retries: 1/10 Probed points range: 7.034770 tolerance: 0.015000

how does one make it do a normal retract QGL as i know that works

vvuk, 2025-06-08:
Was everything working for you before you installed eddy-ng? Because looking at the log in order:

```
Gantry-relative probe points:
0: -0.742104 1: -0.389908 2: 0.294423 3: 0.403197
Actuator Positions:
z: -2.688653 z1: 0.202564 z2: 0.074518 z3: 2.500769
Making the following Z adjustments:
stepper_z = 2.710952
...
Retries: 0/10 Probed points range: 1.145301 tolerance: 0.015000
```
that looks fine...

```
Gantry-relative probe points:
0: 1.172558 1: 0.116493 2: -0.017486 3: -0.043634
Retries: 1/10 Probed points range: 1.216192 tolerance: 0.015000
Actuator Positions:
z: 4.522086 z1: -3.086954 z2: 1.954267 z3: -2.920435
...
Retries: 1/10 Probed points range: 1.216192 tolerance: 0.015000
```
but now it looks like the adjustment moved things in the right direction, but the range is increasing. I'm surprised that QGL isn't aborting sooner due to range increasing. What printer is this?

Platidors, 2025-06-09:
It was about the same when I calibrated on a hot bed and with a hot nozzle. I couldn't do it on a cold bed z-tilt,  I calibrated on a table temperature of 60 degrees and everything started working correctly.

vvuk, 2025-07-16:
Any new info here? Hard to diagnose otherwise.

Jlively1981, 2025-07-17:
I am having a similar issue. I am starting my printer up after a month or two of no use. 

It was homing, tapping, qgl'ing, and printing great before. I did update several softwares when starting back up.

vvuk, 2025-07-27:
Huh. Do you happen to know what version of of klipper you updated from/to? (You can run

cd ~/klipper
git reflag

and copy paste the output here)

mans0ry, 2025-07-29:
I also have same kind of issue, re-installed my head to dragonburner (changed offset y_offset: 31) (previously vanilla + low mount), G28 XYZ working great. When doing QGL nozzle went through the bed on the far right side of the portal

mans0ry, 2025-08-03:
> I also have same kind of issue
My bad, I fixed it on my end. Wasn't any software issues

borinet, 2025-10-08:
so what was the problem?

vvuk, 2026-11-01:
Several of you fixed this by recalibrating after a Klipper update. Older eddy-ng stored calibrations in a format tied to the installed numpy version. `develop` stores them as plain JSON, so updates can't silently break them.

closed 2026-11-01 by vvuk