vvuk/eddy-ng #68: z_offset: manually tune, and different ones for different beds

unlucio, 2025-07-05:
Thank you for this module, perhaps Iìll be finally able to use the eddy probe.

I have an Eddy Coil (I2C)
[image]
connected via my Npatune 4 Pro via a Pi Pico Zero:
[image]

One thing tho: even tho `PROBE_EDDY_NG_SETUP` works fine, I keep ending up with a too low z-offset.
[image]

I'm fine to manually tune it, but how?
In the automatic saving part at the bottom of printer.cfg there's no mention of z_offset:

```toml
[probe_eddy_ng eddy_probe]
calibrated_drive_currents = 15, 16
calibration_version = 5
reg_drive_current = 15
tap_drive_current = 16
calibration_15 = gASVygMAAAAAAAB9lCiMAXaUSwWMBGZ0b2iUjBtudW1weS5wb2x5bm9taWFsLnBvbHlub21pYWyUjApQb2x5bm9taWFslJOUKYGUfZQojARjb2VmlIwVbnVtcHkuY29yZS5tdWx0aWFycmF5lIwMX3JlY29uc3RydWN0lJOUjAVudW1weZSMB25kYXJyYXmUk5RLAIWUQwFilIeUUpQoSwFLCoWUaAyMBWR0eXBllJOUjAJmOJSJiIeUUpQoSwOMATyUTk5OSv////9K/////0sAdJRiiUNQZRJm45k3+T/9n+N62n36Px7kyQEYM+c/vMfalI1n2j9lnvPg8dPGPwlFfkuzEbY/aD2JDPgPxj/fnm08fdPEP0kBSwLyO6I/hXdKR3qNgb+UdJRijAZkb21haW6UaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxACzF/PdG+UPr9y2VdW9JQ+lHSUYowGd2luZG93lGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGKMB19zeW1ib2yUjAF4lIwGc3ltYm9slGgsdWKMCWZ0b2hfaGlnaJRoBSmBlH2UKGgIaAtoDksAhZRoEIeUUpQoSwFLCoWUaBiJQ1Arzqq1qSwaQIpFzox4LAhAJib6Rc/0+T8vPz3zPi75P/KLb6bcH/M/g2ikEV/W8r+imTPS587pv8gt9aX7UQJAz9DWwbd59T88oLs8t9XUv5R0lGJoHWgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQDZiHXOfvlD5nfiYexwuVPpR0lGJoJGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGJoK2gsaC1oLHVijARodG9mlGgFKYGUfZQoaAhoC2gOSwCFlGgQh5RSlChLAUsKhZRoGIlDUGT8JHLY05Q+asmg1mhiKj5YR3U4docdviVs+tsqkAg+HV6oogAx+L2Suvhh2u3lPWZM4kVd6+Y93/VEiopz3b0ewonpRiDUvQWUHGSs7Lw9lHSUYmgdaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxBgTdViJkPYP9ZWZWi5+RNAlHSUYmgkaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxAAAAAAAADwvwAAAAAAAPA/lHSUYmgraCxoLWgsdWKMB2hfcmFuZ2WUXZQoRz/YQyZi1U1gR0At//6/8lMcZYwHZl9yYW5nZZRdlChHQUhT4TPeKABHQUkN+bR+6ABljAJkY5RLD3Uu
calibration_16 = gASVygMAAAAAAAB9lCiMAXaUSwWMBGZ0b2iUjBtudW1weS5wb2x5bm9taWFsLnBvbHlub21pYWyUjApQb2x5bm9taWFslJOUKYGUfZQojARjb2VmlIwVbnVtcHkuY29yZS5tdWx0aWFycmF5lIwMX3JlY29uc3RydWN0lJOUjAVudW1weZSMB25kYXJyYXmUk5RLAIWUQwFilIeUUpQoSwFLCoWUaAyMBWR0eXBllJOUjAJmOJSJiIeUUpQoSwOMATyUTk5OSv////9K/////0sAdJRiiUNQgzYepvz88j8a1HikbPn5P0dtXX0Hruk/0ZZtGE322D85E5QnlwDEPyxkjOuB9uM/cIh3RgJm0j9xq+X85yzovyvZ5txlK7I/XRlyuHZy4z+UdJRijAZkb21haW6UaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxDzVqckZkOUPtOvP5ZsD5U+lHSUYowGd2luZG93lGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGKMB19zeW1ib2yUjAF4lIwGc3ltYm9slGgsdWKMCWZ0b2hfaGlnaJRoBSmBlH2UKGgIaAtoDksAhZRoEIeUUpQoSwFLCoWUaBiJQ1CoC6awzjwWQNP7sOqvJPU//f0nd65H1D/OcN2vMvWzvzczGxQMZMc/sgVAs76L7T/ZIMtkrbbGv/WdapAu5vS/BE4gNWlQrT8se/zMH3jiP5R0lGJoHWgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQNItm0zkKlT78tBjILx6VPpR0lGJoJGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGJoK2gsaC1oLHVijARodG9mlGgFKYGUfZQoaAhoC2gOSwCFlGgQh5RSlChLAUsKhZRoGIlDUAZNrZKQ45Q+AdeRspvMMj5mHNBophsqviAomfG67SI+9SbQ9/GmAj7MlMoYGUcrvlTdyf96CSe+SOtJvEX7OD7HCEqhm84dPnpXp24o2iu+lHSUYmgdaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxDAltzORNBRP9Yt4JiD+BNAlHSUYmgkaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxAAAAAAAADwvwAAAAAAAPA/lHSUYmgraCxoLWgsdWKMB2hfcmFuZ2WUXZQoRz9R0ETO3JbAR0AdkJ0ieaQ2ZYwHZl9yYW5nZZRdlChHQUg+rEFOQABHQUlEcz9RgABljAJkY5RLEHUu
```
Decoding the base 64 reveals a numpy array, but it's binary, so I'm not gonna start touching it.

I also would like to be able to select different z_offsets based on GCODE parameters, but I need to understand where the z-offset is now 😄

thank you again for the awesome module.

vvuk, 2025-07-16:
This is the purpose of tap_adjust_z -- if you set `tap_adjust_z: 0.095` it should behave identically as if you had it set to 0.0 but then did a `SET_GCODE_OFFSET` like you did (after tap). Not sure why 0.030 would be putting the nozzle too high, it should be just adding that amount. Make sure you've updated to the latest code just in case.

In case it helps with any macros you want to do, you can also set tap_adjust_z via `PROBE_EDDY_NG_SET_TAP_ADJUST_Z VALUE=0.095`. It'll take effect on the next TAP.

unlucio, 2025-07-19:
Manually applying the tap_adjust_z, save, and rebbot wasn't a great method to fine-tune this z-offset.
(perhaps tap_adjust_z
Luckily, ModBot happened to make a video on this module https://www.youtube.com/watch?v=ImOpsdSRxt8, and since he had my same issue, his methodology inspired me to find a better way for my setup.

After a series of:
- move the head to the center of the bed
- `PROBE_EDDY_NG_TAP MAX_SAMPLES=<number> TARGET_Z=<number>`
- `PROBE_EDDY_NG_SET_TAP_ADJUST_Z VALUE=<number>`
- move the head **back** to the center of the bed (btw, why does it drift the head away from the point where it probed once the probing is done?)
- put a piece of paper don't he bed
- set z to 0
- move paper and verify friction
- rinse and repeat until happy with the paper friction

I got a good z-offset!

Perhaps this should be an automatic procedure like `PROBE_EDDY_NG_CALIBRATE `, a `PROBE_EDDY_NG_Z_OFFSET_TUNE` that allows for further fine tuning in a more practical way than what I did.

Unfortunately, with my dyslexia, I find Python extremely hard to read (the lack of braces that clearly define blocks makes it very hard for my brain to latch onto the code structure, and the significant white spaces make the whole experience extremely frustrating), but I'd like to contribute a PR on this if it makes sense, tho I never touched klipper's code before so I might need someone to point me int he right direction 😅

This is my current config now:
```
[mcu eddy_bridge]
serial: /dev/serial/by-id/usb-Klipper_rp2040_E6647C7403607D23-if00
restart_method: command

[probe_eddy_ng eddy_probe]
sensor_type: btt_eddy
i2c_mcu: eddy_bridge
i2c_bus: i2c0a
x_offset: 25.4
y_offset: -26.0
tap_max_samples: 20
tap_target_z: -0.550

#*# [probe_eddy_ng eddy_probe]
#*# calibrated_drive_currents = 15, 16
#*# calibration_version = 5
#*# reg_drive_current = 15
#*# tap_drive_current = 16
#*# calibration_15 = gASVygMAAAAAAAB9lCiMAXaUSwWMBGZ0b2iUjBtudW1weS5wb2x5bm9taWFsLnBvbHlub21pYWyUjApQb2x5bm9taWFslJOUKYGUfZQojARjb2VmlIwVbnVtcHkuY29yZS5tdWx0aWFycmF5lIwMX3JlY29uc3RydWN0lJOUjAVudW1weZSMB25kYXJyYXmUk5RLAIWUQwFilIeUUpQoSwFLCoWUaAyMBWR0eXBllJOUjAJmOJSJiIeUUpQoSwOMATyUTk5OSv////9K/////0sAdJRiiUNQNCd9r4fV9z97TBwB27v6Pyjyt606q+Y/tvo+OVHK2D+58Hy3EEjVP/gF3Pww3s4/zUUct4E9ur9rxMCfPra6v+99wCGmEMk/XVnefkIawj+UdJRijAZkb21haW6UaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxANAyhCB2qUPoGirKZO85Q+lHSUYowGd2luZG93lGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGKMB19zeW1ib2yUjAF4lIwGc3ltYm9slGgsdWKMCWZ0b2hfaGlnaJRoBSmBlH2UKGgIaAtoDksAhZRoEIeUUpQoSwFLCoWUaBiJQ1BNxIDr+CYaQOtSmIDmWghAeEI/RWEG+T8B3vm/JZXvP0Q33k3igvU/dQehc3037D+51izWIKjxvz7wP2PnUuS/u0HnY0029z/sJV3vfgrwP5R0lGJoHWgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQZwMEAE3vlD53Q++JxAqVPpR0lGJoJGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGJoK2gsaC1oLHVijARodG9mlGgFKYGUfZQoaAhoC2gOSwCFlGgQh5RSlChLAUsKhZRoGIlDUN+jwv5N0pQ+9n3btNcIKz6tgTL2JsMevqiljVij+g0+JBKCvrvK9r3fRf/kYIfvvRhK08TpYOA9LpeDDj6c+j1QZVSm7cbSvXP6+2pSn+i9lHSUYmgdaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxBgVad/DVfSP1ZFyKoH7RNAlHSUYmgkaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxAAAAAAAADwvwAAAAAAAPA/lHSUYmgraCxoLWgsdWKMB2hfcmFuZ2WUXZQoRz/SVw1/p1VgR0At////////ZYwHZl9yYW5nZZRdlChHQUhVDCg7UABHQUkUox7gMABljAJkY5RLD3Uu
#*# calibration_16 = gASVygMAAAAAAAB9lCiMAXaUSwWMBGZ0b2iUjBtudW1weS5wb2x5bm9taWFsLnBvbHlub21pYWyUjApQb2x5bm9taWFslJOUKYGUfZQojARjb2VmlIwVbnVtcHkuY29yZS5tdWx0aWFycmF5lIwMX3JlY29uc3RydWN0lJOUjAVudW1weZSMB25kYXJyYXmUk5RLAIWUQwFilIeUUpQoSwFLCoWUaAyMBWR0eXBllJOUjAJmOJSJiIeUUpQoSwOMATyUTk5OSv////9K/////0sAdJRiiUNQg07SrhXO8j91gXdkgW36P85mlYWzo+k/GnYvnxdt3j/E8uDbPhXRP1m5Wogi2s8/6exkIw4ggD81BJEQwMLEv/Bdb04VQc8/oEOWU+i30T+UdJRijAZkb21haW6UaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxBqvltX/FuUPm9AJQQbD5U+lHSUYowGd2luZG93lGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGKMB19zeW1ib2yUjAF4lIwGc3ltYm9slGgsdWKMCWZ0b2hfaGlnaJRoBSmBlH2UKGgIaAtoDksAhZRoEIeUUpQoSwFLCoWUaBiJQ1DipXRjtx4aQJIHqrJEJAhAHQlgBYfQ+D/dzdHJU7LsP569TO+6v/M/uN7hIW758j+lj0nsd6LtvwM8pGQSwPC/FaSVEgl39T/pOzbluXfyP5R0lGJoHWgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQ9pS/aLYKlT4z1k/eWieVPpR0lGJoJGgLaA5LAIWUaBCHlFKUKEsBSwKFlGgYiUMQAAAAAAAA8L8AAAAAAADwP5R0lGJoK2gsaC1oLHVijARodG9mlGgFKYGUfZQoaAhoC2gOSwCFlGgQh5RSlChLAUsKhZRoGIlDUDewfNEh6JQ+0Xlp2kCYMD5bNRhyO3okvvXb63nvfxQ+xbvZ07PbBr45Zgd6MHzsPeZLJVYoJrK9JvM/69HO4z2TyxcBSNjbPbN2IB2mhOi9lHSUYmgdaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxAA8mIEyzorP1YVywRQ8BNAlHSUYmgkaAtoDksAhZRoEIeUUpQoSwFLAoWUaBiJQxAAAAAAAADwvwAAAAAAAPA/lHSUYmgraCxoLWgsdWKMB2hfcmFuZ2WUXZQoRz8rOssEYvIAR0At/w54/u66ZYwHZl9yYW5nZZRdlChHQUg0KjNcGABHQUkl77GNOABljAJkY5RLEHUu
#*# tap_adjust_z = -0.02
```
(so I guess that's why 0.03 was too high 😆)
and it's working quite nicely. I've been printing on and off for the whole afternoon, half an hour prints with CAD tinkering in between, and as for now, the heat-creep problem hasn't shown up yet.

An interesting part is:
```
tap_max_samples: 20
tap_target_z: -0.550
```
Despite having an Eddy Coil, these settings are allowing me to use it for homing as well!

I also bought an Eddu DUO (should be delivered this week), because of the temperature drift thing, but since the ModBot guy still seems to have my same issue with an Eddy USB, idk if I should expect improvements or not. Perhaps I might be able to get rid of `tap_max_samples` and `tap_target_z`.
I'll see what happens with that one and perhaps report back.

unlucio, 2025-07-22:
Got the Eddy DUO, updated Kipper, updated eddy-ng, updated firmwares.

Results:
- still got a low z-effset, but less severe than with the Coil
- the same procedure that allowed me to fix the offset before gave me `tap_adjust_z = 0.082` this time,
- got significantly less `!! EDDYng: Tap failed: stddev 0.021 > 0.020. Consider adjusting tap_samples, tap_max_samples, or tap_samples_stddev. !! Tap failed than with the Coil, yet I still got them.
- still have to add `tap_max_samples` and `tap_target_z`, the latter one I set to -0.650 this time
- I then proceeded with a test print (155X15X0.2) but I had to cancel it becasue the z was still too lo and the material was being relaly pushed into the bed, and the nozzle scraping it.
- set the tap `tap_adjust_z=0.1` and then `tap_adjust_z=0.2` but the nozzle still scraped the bed, so I just turned it off because I guess the heat at this pont did some number(?) 

I got a bit of:
```
!! Error during homing stepper_z: Communication timeout during homing
!! Error during homing stepper_z: Communication timeout during homing
!! Error during homing stepper_z: Communication timeout during homing
// idle_timeout: Timeout set to 2100.00 s
!! SDCARD_RESET_FILE cannot be run from the sdcard
!! SDCARD_RESET_FILE cannot be run from the sdcard
```

I'll try again tomorrow, but as for now, the Eddy DUO seems exactly like the Coil.

vvuk, 2026-11-01:
Two additions since this was opened. After a tap, you can babystep and press the "save Z offset" button in Mainsail/Fluidd (`Z_OFFSET_APPLY_PROBE`), and that updates `tap_adjust_z`. On `develop`, tap profiles can hold a `tap_adjust_z` per material. On why the head moves after tapping: tap moves the sensor over the tap point to measure its offset at that temperature, which bed meshes then use.

closed 2026-11-01 by vvuk