vvuk/eddy-ng #74: Inconsistent Z Offset everytime the printer is restarted.

lar213, 2025-07-20:
The z offset is about 0.2mm too low every time the printer is restarted. Have to redo the tap calibration every time.

vvuk, 2025-07-20:
What do you mean by "redo tap calibration"?

lar213, 2025-08-02:
I did some experiments with it and it seems 5 min heat soak helps as well as a SET_GCODE_OFFSET Z_ADJUST=+0.175 for PETG and ABS/ASA and no adjustment for PLA in a filament gcode. I believe the issue is simply the difference between heat expansion in the bed and heat drift in probe makes it off by about 0.2 between the bed temp at 60C and 80C

unlucio, 2025-08-02:
> [@unlucio](https://github.com/unlucio) here https://github.com/vvuk/eddy-ng/wiki#questions-discussion-help

thank you very much