vvuk/eddy-ng #86: Internal error on command:"ACCEPT"

netwolfx, 2025-08-17:
I am running into the following issue when trying to set up the BTT Eddy with Eddy_NG.

Printer mainboard: BTT SKR E3 Mini V2
Printer firmware: Kalico v0.12.0-715-g91fd6480 

Eddy Sensor: BTT Eddy
FIrmware: Kalico v0.12.0-715-g91fd6480 

I can start the probe setup process with PROBE_EDDY_NG_SETUP, and it does start the setup process.
I get to setting the zero point with the paper test, and then I get the above error message and Klipper shuts down.
Attached is an excerpt from the klippy log that shows the errors.
I have also attached the complete log as well.

[Klippy_log_excerpt.txt](https://github.com/user-attachments/files/21823590/Klippy_log_excerpt.txt)

[klippy.log](https://github.com/user-attachments/files/21823594/klippy.log)

I do not know if there is something I'm doing wrong?

3Dbert, 2025-08-26:
I’m seeing what looks like the same shutdown during Eddy_NG setup on a Sovol SV08.

Hardware:
- Mainboard: Sovol SV08 64-bit Silent Mainboard (TMC2209)
- Probe: BTT Eddy DUO
- Host: integrated on the SV08 mainboard (no external SBC)

Firmware:
- Klipper: v0.13.0-229-g3aadda6f-dirty
- Eddy firmware: v0.13.0-229-g3aadda6fb-dirty

What happens:
- Start `PROBE_EDDY_NG_SETUP`
- Do the paper test to set Z=0 manually
- After I finish the manual adjust, the printer lifts Z, then slowly moves down until the nozzle touches the build plate, and *then* Klipper shuts down.
- The UI shows `Internal error on command:"ACCEPT"`, but the failure actually occurs right before/around that final move.

Status / logs:
- `PROBE_EDDY_NG_STATUS`: `Last coil value: 3230442.75 (-infmm) raw: 0x44ea865 (Not calibrated) status: 0x48 UNREADCONV1 DRDY`
- Drive current tested 14–17. At 16 I get: `valid height: 0.004 to 14.999, freq spread 2.08%, Fit 0.0081`.
- `PROBE_EDDY_NG_CALIBRATE` also triggers the same shutdown on my setup.

Minimal repro steps:
1) `FIRMWARE_RESTART` → `RESTART`
2) `G28 X Y`
3) Move to probe point (center)
4) `PROBE_EDDY_NG_SETUP`
5) Complete paper test → printer raises, descends slowly to contact → shutdown (UI then shows ACCEPT error)

I attached the full `klippy.log`. Happy to run any debug build or extra commands.

[klippy.log](https://github.com/user-attachments/files/21994529/klippy.log)

vvuk, 2025-08-26:
Thanks, I'll take a look ASAP -- at first glance this looks like some kind of issue with either python version or numpy version, or an incorrectly installed version of numpy. Can you run `~/klippy-env/bin/pip list` and paste the output here?

netwolfx, 2025-08-27:
Here is the output of  my `~/klippy-env/bin/pip list`:

```pi@neptune2:~ $ ~/klippy-env/bin/pip list
Package           Version
----------------- -------
cffi              1.15.1
greenlet          2.0.2
jinja2            3.1.5
MarkupSafe        2.1.5
msgpack           1.1.0
numpy             2.0.2
packaging         24.2
pip               20.3.4
pkg-resources     0.0.0
pycparser         2.22
pyserial          3.4
python-can        4.5.0
setuptools        44.1.1
typing-extensions 4.13.2
wheel             0.34.2
wrapt             1.17.2
```

I'll note I was getting this same error when running the latest mainline Klipper as well.

3Dbert, 2025-08-27:
In my case this turned out to be caused by the Plotly version in my klippy-env.

I was initially on plotly==6.3.0 and consistently hit the ACCEPT error / shutdown during Eddy-NG setup.
After downgrading to plotly==5.24.1 and doing a full reboot, the setup completes normally and Z-homing with Eddy works as expected.

So if anyone else sees this: check the Plotly version inside ~/klippy-env – going back to 5.24.1 + reboot fixed it for me.

~/klippy-env/bin/pip install "plotly==5.24.1"
sudo reboot

netwolfx, 2025-08-27:
Interesting, I could have sworn that I installed plotly and scipy per the instructions, but clearly it seems I have not.
I will install the 5.24.1 version of plotly, and the latest of scipy, as described in the above comment and report back.

netwolfx, 2025-08-30:
Okay, so I can confirm that downgrading to Plotly version 5.24.1 does resolve the issue and I am able to complete calibration of the BTT Eddy.
Thank you to @3Dbert for finding that fix.

closed 2025-08-30 by netwolfx

vvuk, 2025-08-31:
Ah interesting, thanks for tracking that down. I'll note that in the docs and figure out what's wrong with newer versions