vvuk/eddy-ng #92: error: 'PRODUCT_LDC1612_INTERNALL_CLK' undeclared

snelweg, 2025-09-01:
Just did a fresh git pull in ~/eddy-ng and then a make menuconfig, make

> biqu@SV08:~/klipper$ make clean 
  Creating symbolic link out/board
biqu@SV08:~/klipper$ make menuconfig 
Loaded configuration '/home/biqu/klipper/.config'
No changes to save (for '/home/biqu/klipper/.config')
biqu@SV08:~/klipper$ make
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_icm20948.o
  Compiling out/src/sensor_hx71x.o
  Compiling out/src/sensor_ads1220.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_ldc1612_ng.o
src/sensor_ldc1612_ng.c: In function 'config_ldc1612_ng':
src/sensor_ldc1612_ng.c:352:10: error: 'PRODUCT_LDC1612_INTERNALL_CLK' undeclared (first use in this function); did you mean 'PRODUCT_LDC1612_INTERNAL_CLK'?
     case PRODUCT_LDC1612_INTERNALL_CLK:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          PRODUCT_LDC1612_INTERNAL_CLK
src/sensor_ldc1612_ng.c:352:10: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:65: out/src/sensor_ldc1612_ng.o] Error 1

Steven-Pierce, 2025-09-06:
I've just tried this again today after it failing last weekend and I think it is the same error.
$ biqu@SV08:~/klipper$ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_icm20948.o
  Compiling out/src/sensor_hx71x.o
  Compiling out/src/sensor_ads1220.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_ldc1612_ng.o
src/sensor_ldc1612_ng.c: In function 'config_ldc1612_ng':
src/sensor_ldc1612_ng.c:352:10: error: 'PRODUCT_LDC1612_INTERNALL_CLK' undeclared (first use in this function); did you mean 'PRODUCT_LDC1612_INTERNAL_CLK'?
     case PRODUCT_LDC1612_INTERNALL_CLK:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          PRODUCT_LDC1612_INTERNAL_CLK
src/sensor_ldc1612_ng.c:352:10: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:65: out/src/sensor_ldc1612_ng.o] Error 1
biqu@SV08:~/klipper$

I'm not sure if

vvuk, 2026-11-01:
Fixed, and a `git pull` picks it up.

closed 2026-11-01 by vvuk