vvuk/eddy-ng #94: Issues calibrating probe for tap

nathanprins, 2025-09-04:
First of all, thank you for developing this software and being so helpful in diagnosing issues! 

I am trying to calibrate Eddy NG tap on my Ender 3 using the BTT Eddy Duo, but I'm facing issues with reliability. I am able to run `SETUP` and get a drive calibration for drive current 15. When performing `PROBE_ACCURACY` it yields quite accurate results. However, when attempting to tap, the tap registration is very inconsistent and dependent on temperate of the bed.

I have a feeling the probe accuracy in the range closest to the bed is not good and it's falsely reporting higher values suddenly, causing the tap algorithm to give a false positive.

Here are some successful taps at temperature: 

[image]

[image]

[image]

This is at the same temperature as calibrated. I feel like the hump and jitter at the end of the Z trace should not be there. Do you have any idea how to fix this?

nathanprins, 2025-09-04:
Additional information: No matter which height I put the probe and no matter which drive current I try, every probe reading is quite accurate until I reach within 0.1mm of the bed at which point the reading goes to 552651295.948mm for example

dlhenke, 2025-09-06:
I'm also having the same problem. 
But in my case I have configured eddy ng with bed temp of 100º C and 150º Nozzle temp..
When I print with doors open on my voron the Tap is correct measured. 
But with doors closed and heated chamber printing ASA or ABS or Nylon Tap is done with no success. 

20:13
Tap failed
20:13
Tap failed
20:13
Tap failed
20:13
EDDYng: Tap failed: 3 errors, last: Error during homing probe: Probe completed movement before triggering. If this is a tap, try lowering TARGET_Z or adjusting the THRESHOLD. at toolhead z=-0.250
20:13
Tap 5: z=0.259
20:13
Tap 4: failed (Error during homing probe: Probe completed movement before triggering. If this is a tap, try lowering TARGET_Z or adjusting the THRESHOLD.)
20:13
EDDYng: Tap failed with Z at -0.250: Error during homing probe: Probe completed movement before triggering. If this is a tap, try lowering TARGET_Z or adjusting the THRESHOLD.
20:13
Tap 3: failed (Error during homing probe: Probe completed movement before triggering. If this is a tap, try lowering TARGET_Z or adjusting the THRESHOLD.)
20:13
EDDYng: Tap failed with Z at -0.250: Error during homing probe: Probe completed movement before triggering. If this is a tap, try lowering TARGET_Z or adjusting the THRESHOLD.
20:13
Tap 2: z=0.257
20:13
Tap 1: failed (Error during homing probe: Probe completed movement before triggering. If this is a tap, try lowering TARGET_Z or adjusting the THRESHOLD.)
20:13
EDDYng: Tap failed with Z at -0.250: Error during homing probe: Probe completed movement before triggering. If this is a tap, try lowering TARGET_Z or adjusting the THRESHOLD.

dlhenke, 2025-09-06:
I found a solution ..  vvuk also ran ih the same problem. 
Here is his solution 
https://github.com/vvuk/eddy-ng/pull/82#issuecomment-3184509329

my config now .. 
[probe_eddy_ng btt_eddy]
sensor_type: btt_eddy
i2c_mcu: eddy
i2c_bus: i2c0f
x_offset: 0 # INSERT VALUE FOR YOUR PROBE POSITION
y_offset: 21.47 # INSERT VALUE FOR YOUR PROBE POSITION 
#samples_per_second: 600
tap_target_z: -1
#tap_mode: butter
#tap_threshold: 250
#tap_samples: 3
tap_max_samples: 10
#tap_samples_stddev: .025
#reg_drive_current: 15
#tap_drive_current: 16
tap_start_z: 3.0
##calibration_z_max: 5

And it's working .

limestone-xyz, 2025-10-06:
same issue, especially even worse when I preheat the chamber. (Eddy Duo).

Any ideas - Support?

Tapchicoma, 2026-02-08:
Changing the tap speed to 5 fixed all the problems for me with getting tap values:
```
tap_speed: 5.0
```